LibreHardwareMonitor/LibreHardwareMonitor #1335: Will compensate for fix - Selecting GPU Memory under “Load” tab also directly modifies GPU bus directly under it.

bfrancom22, 2024-05-22:
If anyone can fix this for me I’d be happy to throw you money for your time. I’m sure it’s an easy fix. GPU Memory and GPU Bus  (both under load tab) seem to act as a pair when they shouldn’t. If I rename either one, the other takes the same name automatically upon closing and reopening the program. If I try to add GPU Memory to the gadget, GPU Bus comes with it. I don’t want GPU bus. Upon going back into Libre, both will have the gadget option selected even though I didn’t select it for both. I think there’s some sort of identity issue and both are using the same identifier in the code somewhere.

Taskwi, 2024-07-13:
hi，I have fixed it and share it with you
[Uploading libre hardware monitor.zip…]()

SoggedFox, 2024-10-18:
Same here too with my Gigabyte Gaming OC RTX 4090.

Artem259, 2024-12-14:
Same error. Will this bug be fixed soon? Because GPU Memory parameter is very important for monitoring and in the current state the application is almost unusable.