LibreHardwareMonitor/LibreHardwareMonitor #1352: No CPU Temperature reading on Intel Core Ultra 7 155H

zhongjun96, 2024-06-07:
[LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/15688489/LibreHardwareMonitor.Report.txt)
[image]

Z3r10, 2024-06-13:
I had the same problem (Surface Pro 7), Temp / Freq stopped working (I had it before though)
It was due to Windows Defender blocking the loading of the driver (LibreHardwareMonitor.sys).

Try to add LibreHardwareMonitor folder to Exclusions:

[image]

zhongjun96, 2024-06-13:
> 我遇到了同样的问题（Surface Pro 7），Temp / Freq 停止工作（我以前遇到过这个问题） 这是由于 Windows Defender 阻止了驱动程序（LibreHardwareMonitor.sys）的加载。
> 
> 尝试将 LibreHardwareMonitor 文件夹添加到排除项：
> 
> ![图像](https://private-user-images.githubusercontent.com/142105919/339263775-e04fb3d5-4026-4fa5-8c78-2a2fbba2bc18.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgyNzM2MTksIm5iZiI6MTcxODI3MzMxOSwicGF0aCI6Ii8xNDIxMDU5MTkvMzM5MjYzNzc1LWUwNGZiM2Q1LTQwMjYtNGZhNS04Yzc4LTJhMmZiYmEyYmMxOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYxM1QxMDA4MzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mNWQ2NmU3YTY3NDk3Nzg5NTFmNWRkNThkMWUxOGZiZDVjYTAzZTUwZjQxZjc3MmUwYTQ0NjJkZTkwMWE2MDk3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.tA6SW8CPfq5wpQU93wIZsV52yLSIPGMc-wa6rFgApys)

Thanks for your reply, but I don't have Windows Defender enabled.

zhongjun96, 2024-06-13:
> I had the same problem (Surface Pro 7), Temp / Freq stopped working (I had it before though) It was due to Windows Defender blocking the loading of the driver (LibreHardwareMonitor.sys).
>
> Try to add [...]

Thanks for your reply, but I don't have Windows Defender enabled.

PaulLesur, 2024-07-29:
I have the same issue, adding the folder to Windows Defender Exclusions does not resolve the issue.

qychen2001, 2024-10-06:
My CPU is an Ultra 9 185H and I had a similar problem at first. This is because the latest stable version does not support meteor lake. You can download the latest nightly build version from [here](https://nightly.link/LibreHardwareMonitor/LibreHardwareMonitor/workflows/master/master/LibreHardwareMonitor-net472.zip).
This worked for my system and CPU version and no additional setup was required.

zhongjun96, 2024-10-07:
> My CPU is an Ultra 9 185H and I had a similar problem at first. This is because the latest stable version does not support meteor lake. You can download the latest nightly build version from [...]

Cool, perfect solution

closed 2024-10-07 by zhongjun96

MaHoBo, 2024-10-16:
you can try the code from my pull request. 
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1495