LibreHardwareMonitor/LibreHardwareMonitor #1361: Abnormally High CPU Power, Clock Frequency, and Temperature on Legion R7000P with AMD Ryzen 7 5800H

ab-Royo, 2024-06-15:
I am experiencing abnormally high CPU power, clock frequency, and temperature on my Legion R7000P with an AMD Ryzen 7 5800H CPU. This issue started appearing in the past few days. Before this issue arose, I updated Windows with the following updates:

- KB5037771
- KB5039212

I am not sure if these Windows updates are causing the issue. I have also tested with the latest version of LibreHardwareMonitor, and the issue persists.

### Additional Information:
I have checked previous issues and confirmed that I have not enabled core isolation.
I do not use Windows Defender.

### System Information:
Laptop Model: Legion R7000P
CPU: AMD Ryzen 7 5800H
Windows Updates: KB5037771, KB5039212
[image]
[LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/15845169/LibreHardwareMonitor.Report.txt)

Please let me know if any additional information is needed to diagnose and resolve this issue.

Thank you.

drewaaaaa, 2024-06-15:
KB5039212 for Win10 causes same.

Edmondguo, 2024-06-15:
Same problem, and my cpu is 5800x3d, the Tctl/Tdie temperature is 192C.
I think it's Windows updates that causes this problem. Because two different software on my computer raise the same problem([FanControl](https://github.com/Rem0o/FanControl.Releases) and [TrafficMonitor](https://github.com/zhongyang219/TrafficMonitor)).

ab-Royo, 2024-06-16:
> Win10 的 KB5039211 会导致同样的问题。 删除 KB5039211 毫无意义。 ![活动](https://private-user-images.githubusercontent.com/50954426/339965235-ad1b44a5-66f1-4276-ade1-57cfb15d29c0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg1Mjc3MDAsIm5iZiI6MTcxODUyNzQwMCwicGF0aCI6Ii81MDk1NDQyNi8zMzk5NjUyMzUtYWQxYjQ0YTUtNjZmMS00Mjc2LWFkZTEtNTdjZmIxNWQyOWMwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjE2VDA4NDMyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJlNmFjMDg5YmM1MGYyZjUzYjFhNTg5M2VjZGYwODZiNjc5Yzc2ZmVkNmIzZjczMTE3YWY0ODBjMzA3NzRmZWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4diZRSY3LrH2NY0Wub8OqFq6NOp0HZ25kSO7M9IfhSA)
> 
> 温度可能也是 256 度 🙄

I also tried removing KB5039212 and the problem did not go away

Edmondguo, 2024-06-17:
I'm not sure whether it is system's fault. I uninstall FanControl, and let the BIOS controls the fans in my case. I run stress FPU by AIDA64, and it seems like the BIOS fan control strategy doesn't work, because the the speed of fans doesn't increase as expected. Can Windows system's fault influence the BIOS fan control strategy? Maybe I should try to use PE to stress FPU.

drewaaaaa, 2024-06-22:
I did nothing, but now all showing correct. I'm confused. 🤷🏼‍♂️

AbhiMayadam, 2024-08-25:
That's been a bug since the openhardwaremonitor days at least with power.