LibreHardwareMonitor/LibreHardwareMonitor #1374: Not display CPU Temperature with use dll

CzarOfScripts, 2024-07-07:
I used the latest nightly build and in the GUI everything shows up fine, but when using the dll from there in my program, I can't get the temperature and some other non important stats for me. By the way, this problem does not occur on my computer.

<details>
  <summary>Libre Hardware Monitor GUI</summary>
  
  [image]
</details>

From my own program I generate a report where I get all the information about the sensors and get the following picture:

<details>
  <summary>report.txt</summary>
  
```
|-----------------------
| System Information Report
|
| Used library: LibreHardwareMonitor
|-----------------------

CPU Temperature: 0°C
       CPU Load: 2,489209%

GPU Temperature: 49°C
       GPU Load: 1%

       RAM Load: 29,21803%
  RAM Available: 22,39998 GB
       RAM Used: 9,246471 GB

/----------
|       Cpu
|---------- Load
|      3,43 - CPU Core #1
|      1,15 - CPU Core #2
|      2,82 - CPU Core #3
|      2,46 - CPU Core #4
|      4,59 - CPU Core #5
|      0,82 - CPU Core #6
|      4,43 - CPU Core #7
|      1,68 - CPU Core #8
|      1,65 - CPU Core #9
|      1,18 - CPU Core #10
|      0,96 - CPU Core #11
|      4,69 - CPU Core #12
|      2,49 - CPU Total
|      4,69 - CPU Core Max
|
|---------- Power
|      0,00 - Package
|      0,00 - Core #1 (SMU)
|      0,00 - Core #2 (SMU)
|      0,00 - Core #3 (SMU)
|      0,00 - Core #4 (SMU)
|      0,00 - Core #5 (SMU)
|      0,00 - Core #6 (SMU)
|
|---------- Clock
|  не число - Core #1
|  не число - Core #2
|  не число - Core #3
|  не число - Core #4
|  не число - Core #5
|  не число - Core #6
|
|---------- Factor
|  не число - Core #1
|  не число - Core #2
|  не число - Core #3
|  не число - Core #4
|  не число - Core #5
|  не число - Core #6
|
|---------- Voltage
|      1,55 - Core #1 VID
|      1,55 - Core #2 VID
|      1,55 - Core #3 VID
|      1,55 - Core #4 VID
|      1,55 - Core #5 VID
|      1,55 - Core #6 VID
|
|---------- Temperature
|      0,00 - Core (Tctl/Tdie)
|

/----------
|    Memory
|---------- Data
|      9,25 - Memory Used
|     22,40 - Memory Available
|     14,32 - Virtual Memory Used
|     25,14 - Virtual Memory Available
|
|---------- Load
|     29,22 - Memory
|     36,28 - Virtual Memory
|

/----------
|    GpuAmd
|---------- Factor
|      0,00 - Fullscreen FPS
|
|---------- SmallData
|   1046,20 - D3D Dedicated Memory Used
|    143,53 - D3D Shared Memory Used
|
|---------- Load
|      1,17 - D3D 3D
|      0,00 - D3D Copy
|      0,00 - D3D Compute 0
|      0,00 - D3D Compute 1
|      0,00 - D3D Copy
|      0,00 - D3D Timer 0
|      0,00 - D3D Security 1
|      0,00 - D3D True Audio 0
|      0,00 - D3D Compute 3
|      0,00 - D3D High Priority Compute
|      0,00 - D3D High Priority 3D
|      0,00 - D3D Video JPEG
|      0,00 - D3D Video Decode 1
|      0,00 - D3D True Audio 1
|      0,00 - D3D Video Codec 0
|      0,00 - GPU Core
|      0,00 - GPU Memory
|
|---------- Power
|     33,00 - GPU Package
|
|---------- Temperature
|     49,00 - GPU Core
|     58,00 - GPU Memory
|     47,00 - GPU VR VDDC
|     49,00 - GPU Hot Spot
|
|---------- Clock
|     14,00 - GPU Core
|    239,00 - GPU SoC
|   1748,00 - GPU Memory
|
|---------- Fan
|   1890,00 - GPU Fan
|
|---------- Control
|     54,00 - GPU Fan
|
|---------- Voltage
|      0,73 - GPU Core
|      0,79 - GPU SoC
|      0,85 - GPU Memory
|
```

</details>

The code can be viewed either in my [repository](https://github.com/CzarOfScripts/system-monitoring-cSharp/blob/main/SystemMonitoring/SystemInformationProvider/LibreHardwareMonitorProvider.cs)
<details>
  <summary>... or under a spoiler</summary>
  
```cpp
﻿using System;
using System.IO;
using System.Linq;
using System.Text;

using LibreHardwareMonitor.Hardware;

namespace App
{
	public class LibreHardwareMonitorProvider : SystemInformationProvider
	{
		private readonly Computer _computer;

		public LibreHardwareMonitorProvider()
		{
			_computer = new Computer() { IsCpuEnabled = true, IsGpuEnabled = true, IsMemoryEnabled = true };
			_computer.Open();
			_computer.Accept(new UpdateVisitor());
		}

		public override void SaveReport(string path)
		{
			SystemDataInformation systemData = GetSystemInformation();

			StringBuilder report = new StringBuilder();
			report.AppendLine("|-----------------------");
			report.AppendLine("| System Information Report");
			report.AppendLine("|");
			report.AppendLine("| Used library: LibreHardwareMonitor");
			report.AppendLine("|-----------------------");
			report.AppendLine();

			report.AppendLine($"{"CPU Temperature:",16} {systemData.cpuTemperature}°C");
			report.AppendLine($"{"CPU Load:",16} {systemData.cpuLoad}%");
			report.AppendLine();

			report.AppendLine($"{"GPU Temperature:",16} {systemData.gpuTemperature}°C");
			report.AppendLine($"{"GPU Load:",16} {systemData.gpuLoad}%");
			report.AppendLine();

			report.AppendLine($"{"RAM Load:",16} {systemData.ramLoad}%");
			report.AppendLine($"{"RAM Available:",16} {systemData.ramAvailable} GB");
			report.AppendLine($"{"RAM Used:",16} {systemData.ramUsed} GB");
			report.AppendLine();

			foreach (IHardware hardware in _computer.Hardware)
			{
				hardware.Update();

				report.AppendLine();
				report.AppendLine();
				report.AppendLine("/----------");
				report.AppendLine($"| {hardware.HardwareType,9}");

				var sensorGroups = hardware.Sensors.GroupBy(s => s.SensorType);
				foreach (var group in sensorGroups)
				{
					report.AppendLine($"|---------- {group.Key}");
					foreach (ISensor sensor in group)
					{
						report.AppendLine($"| {sensor.Value,9:F2} - {sensor.Name}");
					}
					report.AppendLine("|");
				}
			}

			File.WriteAllText(path, report.ToString());
			System.Diagnostics.Process.Start(path);
		}

		public override SystemDataInformation GetSystemInformation()
		{
			float cpuTemperature = 0;
			float cpuLoad = 0;

			byte gpuTemperature = 0;
			byte gpuLoad = 0;

			float ramLoad = 0;
			float ramAvailable = 0;
			float ramUsed = 0;

			foreach (IHardware hardware in _computer.Hardware)
			{
				hardware.Update();

				if (hardware.HardwareType == HardwareType.Cpu)
				{
					cpuTemperature = GetSensorValue(hardware, SensorType.Temperature, "Core (Tctl/Tdie)");
					cpuLoad = GetSensorValue(hardware, SensorType.Load, "CPU Total");
				}
				else if (hardware.HardwareType == HardwareType.GpuNvidia || hardware.HardwareType == HardwareType.GpuAmd || hardware.HardwareType == HardwareType.GpuIntel)
				{
					gpuTemperature = Convert.ToByte(GetSensorValue(hardware, SensorType.Temperature, "GPU Core"));
					gpuLoad = Convert.ToByte(GetSensorValue(hardware, SensorType.Load, "GPU Core"));
				}
				else if (hardware.HardwareType == HardwareType.Memory)
				{
					ramUsed = GetSensorValue(hardware, SensorType.Data, "Memory Used");
					ramAvailable = GetSensorValue(hardware, SensorType.Data, "Memory Available");
					ramLoad = GetSensorValue(hardware, SensorType.Load);
				}
			}

			return new SystemDataInformation(cpuTemperature, cpuLoad, gpuTemperature, gpuLoad, ramLoad, ramAvailable, ramUsed);
		}

		private float GetSensorValue(IHardware hardware, SensorType sensorType, string sensorName = null)
		{
			ISensor sensor = hardware.Sensors.FirstOrDefault(s => s.SensorType == sensorType && (sensorName == null || s.Name.Contains(sensorName)));
			return sensor != null ? (float) sensor.Value : 0;
		}

		public void Close()
		{
			_computer.Close();
		}
	}
}
```
</details>

sebastian-dev, 2024-07-08:
Your software must have admin rights or be started with admin rights, otherwise the lib cannot use the driver.

CzarOfScripts, 2024-07-08:
@sebastian-dev, the program was run as admin, it didn't help.