LibreHardwareMonitor/LibreHardwareMonitor #1392: No core temps on amd ryzen 3700x

Benwashear25, 2024-07-30:
When i open the app under the cpu i see "core (Tctl/Tdie)" "SoC" and "CCD1(Tdie) no core temps. All other temps show like gpu core and hotspot and even the temps on my ssd. I tried excluding it from windows defender and it made no difference. If anyone has any ideas on how to see all the temps. 
[image]

RadekJurzysta, 2024-07-30:
+1 for Ryzen 5600x on MSI B550 Tomahawk

fawaf, 2024-07-30:
i believe there is no core temps for any cpu? (or at least amd ones)

don't see it either for my 7800x3d on msi x670e tomahawk

randallroman, 2024-08-01:
5800x3D on ASUS TUF GAMING B550-PLUS WIFI II.  No individual core temps.  Just CCD1 and Core (Tctl/Tidle)