LibreHardwareMonitor/LibreHardwareMonitor #1396: Dark Mode Theme

xmikedanielsx, 2024-07-30:
Request Person: JayzTwoCents
Request From: https://www.youtube.com/watch?v=sX2ThUWNXio
Request:
As an end User using high contrast screens or not, it seems imperative in any current application to be able to utilize a dark mode. As new screens and evolving technologies in them seem to have a higher contrast ratio and brightness levels and this tends to hurt the eyes. Also, there are many other benefits from having dark mode.

Acceptance Criteria:
The application itself will need to be changed to allow for the back and forecolors to be altered depending on in dark mode or not. An easy switcher would need to be implemented. An example Mock Up of the switcher has been made.
[image]

pedrosimoes79, 2024-07-30:
there is anyone already doing this one?

xmikedanielsx, 2024-07-30:
I took a quick stab at it. but please please do it! .

Tellinq, 2024-07-31:
I hate to be _that_ guy…. FYI, this is a duplicate of #116 and PR #847 

I recommend closing this issue in favor of the PR/older issue.

PhyxionNL, 2024-07-31:
Yes, closing due to dupe of #116.

closed 2024-07-31 by PhyxionNL