LibreHardwareMonitor/LibreHardwareMonitor #1398: Only log out checked sensor

xmikedanielsx, 2024-07-30:
Request Person: JayzTwoCents
Request From: https://www.youtube.com/watch?v=sX2ThUWNXio
Request:
As an end user when I log to CSV to only log the current visible sensors and do not log the hidden sensors . Along with a menu item to hold the value of log hidden or not.

Acceptance Criteria:
A log file that contains only Visible sensors and not Hidden Sensors that obeys the option.

xmikedanielsx, 2024-07-30:
Only Concern I have here is if the user change sensors during recording. .. What should be the action? Should we be able to uncheck sensors or check sensors if you're currently logging?

mandel99, 2024-08-19:
I think the best solution is disable changing sensors during logging.