LibreHardwareMonitor/LibreHardwareMonitor #1399: Python bindings please!

BBC-Esq, 2024-07-30:
I'd love to incorporate the metrics that this project can gather in my python programs.  Are there any plans to port it to python and/or make some python bindings whatever the correct terminology is?

GitHub-source-42, 2024-07-30:
it would be trying to make a python library that can be imported for use in other programs like:

From LibreHardwareMonitor import *

I am by no means an expert but I can start having a look from the source code

BBC-Esq, 2024-07-30:
I subsequently came across this old comment:

https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/809

Perhaps making the program able to be run "headless" and/or purely in server mode and/or adding a command line way of running it might be a possible solution as well.  For my purposes, I'd just like to be able to incorporate the metrics that it gathers into a Python program and not have a user worry about starting it up separately.  For example, it seems possible to get the metrics via python pursuant to the above comment, although I haven't tested it, but it's just cumbersome...

PhyxionNL, 2024-08-06:
Not planned, closing.

closed 2024-08-06 by PhyxionNL