LibreHardwareMonitor/LibreHardwareMonitor #1408: System.ArgumentOutOfRangeException: Specified argument was out of the range of valid values.

tomasz1986, 2024-08-02:
I keep getting this error when LibreHardwareMonitor is running in background. It just pops up randomly, and if I click "Continue", the program will not close and keep running, however the same error message will then keep popping up from time to time. This doesn't seem to happen when the program is in foreground.

[image]

```
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.ArgumentOutOfRangeException: Specified argument was out of the range of valid values.
Parameter name: Index out of range
   at Aga.Controls.Tree.TreeViewAdv.ClearNodesSize(TreeModelEventArgs e, TreeNodeAdv parent) in D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeViewAdv.cs:line 1237

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4739.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
LibreHardwareMonitor
    Assembly Version: 0.9.4.0
    Win32 Version: 0.9.4.0
    CodeBase: file:///C:/Users/User/XXX/LibreHardwareMonitor/app/LibreHardwareMonitor.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4734.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4718.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4390.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
Aga.Controls
    Assembly Version: 1.7.0.0
    Win32 Version: 1.7.0.0
    CodeBase: file:///C:/Users/User/XXX/LibreHardwareMonitor/app/Aga.Controls.DLL
----------------------------------------
LibreHardwareMonitorLib
    Assembly Version: 0.9.4.0
    Win32 Version: 0.9.4.0
    CodeBase: file:///C:/Users/User/XXX/LibreHardwareMonitor/app/LibreHardwareMonitorLib.DLL
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4724.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
Microsoft.Win32.TaskScheduler
    Assembly Version: 2.11.0.0
    Win32 Version: 2.11.0.0
    CodeBase: file:///C:/Users/User/XXX/LibreHardwareMonitor/app/Microsoft.Win32.TaskScheduler.DLL
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4190.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Management
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4435.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Management/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Management.dll
----------------------------------------
System.Configuration.Install
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration.Install/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.Install.dll
----------------------------------------
OxyPlot
    Assembly Version: 2.1.2.0
    Win32 Version: 2.1.2.0
    CodeBase: file:///C:/Users/User/XXX/LibreHardwareMonitor/app/OxyPlot.DLL
----------------------------------------
OxyPlot.WindowsForms
    Assembly Version: 2.1.2.0
    Win32 Version: 2.1.2.0
    CodeBase: file:///C:/Users/User/XXX/LibreHardwareMonitor/app/OxyPlot.WindowsForms.DLL
----------------------------------------
Accessibility
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------
System.Management.resources
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Management.resources/v4.0_4.0.0.0_pl_b03f5f7f11d50a3a/System.Management.resources.dll
----------------------------------------
pazsqf5j
    Assembly Version: 0.0.0.0
    Win32 Version: 4.8.4718.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
HidSharp
    Assembly Version: 2.1.0.0
    Win32 Version: 2.1.0.0
    CodeBase: file:///C:/Users/User/XXX/LibreHardwareMonitor/app/HidSharp.DLL
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.
```

I get the error both on the stable https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/releases/tag/v0.9.3 release and the current nightly (as of August 2, 2024).

The OS is Windows 10 Enterprise LTSC 2021 x64.

PhyxionNL, 2024-08-06:
Dupe #688

closed 2024-08-06 by PhyxionNL

MayankJairaj, 2025-03-19:
For now I have just used

MayankJairaj, 2025-03-19:
For now I have just skipped the exception, will try and find some solution. 

if (index < 0 || index >= parent.Nodes.Count)
{
    continue;
}