LibreHardwareMonitor/LibreHardwareMonitor #1410: Don't Start on Boot

corazon008, 2024-08-04:
So as you can see I checked those options
[image]
But I don't see the gadget on the screen nor the icon at the bottom-left.

Additionnaly I tried to create some task with the task scheduler manually but nothing work.

Since this project is a fork of OpenHardwareMonitor, I add that the option in OHM for starting with Windows work.

My config: 2 users, 1 admin, 1 non-admin (my account)

keif888, 2024-08-21:
The tick against "Run On Windows Startup" is populated by checking for a task or a registry key that is created when you click on the menu option.  For it to be ticked indicates that one of these exists.  Clicking on the already ticked menu option will remove the task/registry key.

If you have created a task yourself, and it's got a start program action set correctly to the location of LibreHardwareMonitor.exe, then this will also cause the menu item to show as ticked, BUT it will not be removed when you click the menu option, unless the task name is correct.  
_It is possible that this is the cause of your issue, and you should delete all of the tasks that you created for LibreHardwareMonitor, and then check if the Run on Windows Startup is still ticked._

If the user is an admin, it will use a task (so long as the task scheduler connected), otherwise it will use a user specific registry key.

The task is in the Task Scheduler Library folder within Task Scheduler and called LibreHardwareMonitor.  It is triggered by the logon of any user.

[image]

The registry key is in HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Run and should be called LibreHardwareMonitor with a value equal to the location of the LibreHardwareMonitor.exe file.

corazon008, 2024-09-04:
@keif888 Hello, I removed all the test that I created and untick everything reboot aand then tick everything but it's still don't start

closed 2026-01-03 by PhyxionNL