LibreHardwareMonitor/LibreHardwareMonitor #1411: IT8620E: Only 3 fans getting recognized

d-e-n-n-i-s-t, 2024-08-05:
Hello,

I have a Gigabyte GA-H97-D3H motherboard and wanted to control my fans using FanControl. Unfortunately, only 3 of my 5 fans get recognized, and I also checked for the same behavior in LibreHardwareMonitor. The weird thing is, all 5 fan speeds get recognized, but I only have 3 controls
[image]

Is it possible to somehow add support for all fans? In "LibreHardwareMonitor/LibreHardwareMonitorLib/Hardware/Motherboard
/SuperIOHardware.cs", I saw that there are 2 other motherboards with the same SuperIO chip (IT8620E) with max 2 System Fans. Could one just add 2 more? Or would some kind of mapping somewhere else be required in order to add more fans? I don't know much about the topology of this project. I guess there'd have to be a whole new entry created for my motherboard.

Tellinq, 2024-08-05:
It should be possible to add support for all 5 of your fans.

I can do this for you, but I'm going to need some steps:

I need to know which fan index is applicable to each header (Where is CPU Fan, CPU Fan Optional, System Fan `#1-3`?)

If you want to also get more useful data regarding the other sensors which I'll need for a proper addition of your board, alternative software for comparison (HWiNFO64 is a common app to compare with as that has the biggest chance of having useful data) would be helpful.

For now, I've done the following:

1. Based off the screenshot you sent, I've added detection for that motherboard in a separate data entry
2. Matched the fan and control addition based off the fan length that your board should support. If it still doesn't show the correct amount, I'll try manually adding the sensors next.
3. Added 50 potential temperature indexes. This is just to see if there are any staggering sensors that might exist outside of your typical index. I'll need external software like HWiNFO to give me a good idea of which sensors match. Screenshots of comparison results side by side in the same screenshot would help drastically.
4. Voltages basically remain generic. I'll need external software like HWiNFO to get a good idea of which is which.

If you aren't sure which fan is which, as long as the fans are properly detected now in this build, you should be able to use FanControl by copying `LibreHardwareMonitor.dll` and `LibreHardwareMonitor.pdb` in the dedicated LibreHardwareMonitor folder to FanControl. Just make sure that the .NET version matches. (example if you are using .NET 8 of FanControl, use the .NET 8 folder, and vise versa for .NET 4.8)

Here's the test build I just set up: https://github.com/Tellinq/LibreHardwareMonitor/actions/runs/10257030564. Let me know what results you get.

d-e-n-n-i-s-t, 2024-08-06:
@Tellinq  Wow, thank you for your effort! Alright, I'll try to provide information for each of your inquiries as precise as possible:

> I need to know which fan index is applicable to each header (Where is CPU Fan, CPU Fan Optional, System Fan #1-3?)

The 1st fan is the CPU Fan. The 2nd fan is CPU Opt. Fans 3-5 are system fans. -> 5 fans in total.

> If you want to also get more useful data regarding the other sensors which I'll need for a proper addition of your board, alternative software for comparison (HWiNFO64 is a common app to compare with as that has the biggest chance of having useful data) would be helpful.

I don't really need the rest, however it's not useless and may help. I made a screenshot of HWiNFO64, is this what you need? I also created a [HWiNFO64-report](https://github.com/user-attachments/files/16503754/hwinfoReport_gigabyte-ga-h97-d3h.zip) of motherboard, buses, ports, battery and sensors. Pretty sure that it contains most of the information you need.
Had to pack the report into a zip file for uploading to github.

[image]

> Here's the test build I just set up: https://github.com/Tellinq/LibreHardwareMonitor/actions/runs/10257030564. Let me know what results you get.

I don't really know if I understand what to download there; I use the .NET 4.8 version. When I clicked on your link, I went to the "Artifacts" section. Is that where you want me to download the test build from? I tried running the "LibreHardwareMonitor-net472" artifact, as it seemed to be the closest to .NET4.8 to me. Was that correct?

Running the LibreHardwareMonitor executable in that artifact-zip, I now have a "3V standby" voltage, so something changed. However, I still only have 3 fan controls:

[image]

I don't know if I did what you need me to do here. Maybe give me some feedback on that. Didn't try running FanControl as the fan controls should appear in LibreHardwareMonitor, too (?)

Tellinq, 2024-08-06:
Thanks for that extra info! 

Here's a new build: https://github.com/Tellinq/LibreHardwareMonitor/actions/runs/10258430385

Made a few changes:
- IT8620E should now explicitly define 5 controls instead of the default 3
- All sensors should have appropriate names, so long as I properly defined correctly.
- I've defined the fan names in correspondence to what HWiNFO64 claims to be correct. However, once we get to the point of being able to test each fan control within FanControl, I'd just like confirmation that each fan control that I've named is properly indicated when you attempt to control each (ex System Fan `#2` should be controlling `SYS_FAN2` from your motherboard, and not any other header)

 > I don't really understand what to download there... I use the .NET 4.8 version. When I clicked on your link, I went to the "Artifacts" section. Is that where you want me to download the test build from? I tried running the "LibreHardwareMonitor-net472" artifact, as it seemed to be the closest to .NET4.8 to me. Was that correct?
 
 Yup, that's the correct one!
 
 > I don't know if I did what you need me to do here. Maybe give me some feedback on that. Didn't try running FanControl as the fan controls should appear in LibreHardwareMonitor, too (?)
 
Basically just needing to return back screenshots confirming that the data entry I've set up is correct. Overall, I simply just need verification that each fan control is detected and can be controlled in FanControl. Once the fan controls *do* show up in LHM, then we can move on to verifying that the controls actually are doing something.

So all I just need answered back:
1. A screenshot of LibreHardwareMonitor and HWiNFO64 side by side showing the motherboard data entries. I just need to confirm these align, although I'm confident that they should now.
2. Confirmation that the 5 fan controls are properly detected now.
3. Confirmation via FanControl that the new controls actually *work*, and then confirming that each fan control is actually identified correctly for each header. If not, I'll make sure to correct them.

Tellinq, 2024-08-06:
For perspective, when I say a side-by-side screenshot, I'm referring to a screenshot like this:
[image]

d-e-n-n-i-s-t, 2024-08-06:
@Tellinq Okay! After a bunch of testing, I come here to announce your success!

First of all, the comparison to HWiNFO:

[image]
Seems fine to me.
As seen in the screenshot, LHM now detects all 5 fan speeds + fan controls

The controls also work, and they also correspond to their respective header on the motherboard. I confirmed this by turning a fan to 0% (0 RPM) in LHM, checking which fan turns off visually, then turning the fan back on and physically plugging out the fan header it's supposed to control in LHM. All fan control names correspond to the correct fan header. As to why Gigabyte sorted the System Fans not 1, 2, 3 but 2, 3, 1... I guess we'll never know 😂

Also checked that these controls work when setting up FanControl using the `LibreHardwareMonitor.dll` and `LibreHardwareMonitor.pdb` you provided. Works perfectly 👍🏼

[image]
(all case fans being on 30% is a coincidence here :D They work, tested manually)

Thank you very much. You saved me from being forced to use SpeedFan... :D

d-e-n-n-i-s-t, 2024-08-06:
Will this get merged? Then I wouldn't have to manually inject those files you provided (once FanControl updates its LHM version)

Tellinq, 2024-08-06:
Perfect! I just made some final wording changes for voltage sensors (this won’t change anything in what you see in FanControl). I just created #1413. Once merged, FanControl is typically quick with updating LHM accordingly.

Oh also, I can’t exactly comment on why the System fan # is out of order, but if I were to guess it’s likely due to the physical layout Gigabyte wanted in the headers, and just so happened that the SuperIO chip had to read the order differently. Thankfully with apps such as FanControl that sort by added control instead of by index, it gives me a bit of headroom to reorder accordingly.

I’m glad I saw this issue today as I had a similar board with this SuperIO chip. Thank you very much for testing and I’m glad that I’m able to help another person be able to utilize LHM and any other app that uses it!

closed 2024-08-06 by PhyxionNL