LibreHardwareMonitor/LibreHardwareMonitor #1414: Temperature Minus :D

YuYuOneROG, 2024-08-07:
[image]

Temperatuyre #5 is minus

Tellinq, 2024-08-07:
This is pretty typical for NCT6798D on ASUS boards. I end up submitting PRs to add boards with the proper temperature index

Technically, that’s an auxiliary sensor that is acting more like noise.

I can add your specific board if you’d like with the relevant sources. Procedure would be similar to #1411 for reference

Typically with modern cheap ASUS boards, this will consist with CPU, motherboard, and VRM (not always included, but been more common since Intel 600 series and onward). The rest are typically auxiliary

dtromb, 2025-01-23:
I'm also seeing #5 flying around wildly (from 5C to 188C on an ASUS Prime Z790 - not exactly a "cheap" board.  From what I understand this is an unassigned / not connected aux sensor, the input pin is just floating.   Every other monitoring software package I've tried knows to ignore / not display this.   (I lost about two hours today being concerned with the non-reading from this "sensor" using LHM, oof.)

Would it be super-hard to just not display aux sensors outputs that don't have sensors attached, like HWiNFO and others do?  Might save others the trouble...

Tellinq, 2025-01-23:
> Would it be super-hard to just not display aux sensors outputs that don't have sensors attached, like HWiNFO and others do? Might save others the trouble...

Only reason why these are being displayed is because LHM is pulling data from the SuperIO and the values that it can read. It's not going to know if there's an aux sensor unless it's manually defined in a data entry, at which point will be already chosen to show or not.

Reason why HWiNFO and others may not show these is because they like have a specific-board entry or a broad entry for a platform. Reason LHM doesn't for your particular board is because it wasn't tested for your specific board. That's why in my initial reply I asked for a screenshot comparison between HWiNFO and LHM that way I could potentially add a specific data entry.