LibreHardwareMonitor/LibreHardwareMonitor #1418: AMD Ryzen5 9600X CPU can't get temperature

Quttia, 2024-08-09:


nikos1988, 2024-08-11:
Same here with Ryzen 7 9700X

RMerl, 2024-08-18:
Slight variant on this: I have a Ryzen 9950X here, and I discovered that LibreHardwareMonitor (through FanControl) labels the Tdie temperature as "PCH" instead.  This is based on monitoring the temperatures with HWInfo 64.  Can you folks confirm that you see the same thing?

SakuraMiku27, 2024-08-25:
Same here with Ryzen 9 9950X
[LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/16740483/LibreHardwareMonitor.Report.txt)

GracieSW584, 2024-08-26:
I also cannot monitor my CPU temperatures. AMD Ryzen 9 9900X. 

Interestingly, the older version of HWinfo doesn't work properly with it either, and detects the CPU as an Intel Pentium. 

Seems like something in Zen 5 might have changed enough to create these issues?

definatelynotlhero, 2024-09-03:
the new versions of HWINFO read the sensors of 9950x properly. Is there a way to fix this lib, for it to work also?

Galeforce1, 2024-09-03:
Same here with Ryzen 7 8845HS. Not getting CPU temperature, core voltages and wattage, or core clock speeds.
[LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/16845305/LibreHardwareMonitor.Report.txt)

Galeforce1, 2024-09-10:
> Same here with Ryzen 7 8845HS. Not getting CPU temperature, core voltages and wattage, or core clock speeds. [LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/16845305/LibreHardwareMonitor.Report.txt)

And now it's working again. Not sure what happened, I didn't update the app - or anything.

JohnWayne77, 2024-09-17:
I am also having trouble getting CPU temp reading on a 9600X. I guess that this issue applies to all the AMD 9000 Series CPUs.

closed 2024-09-20 by PhyxionNL