LibreHardwareMonitor/LibreHardwareMonitor #1424: MB Vcore Doubled

STRETCHDEATH, 2024-08-13:
Noticed mb vcore voltage on z490 taichi (nuvoton) was exactly double my shown vcore in hardware monitor and cpu-z, I understand the cpu section has voltages but they are vid... when running an undervolt that is irrelevant because you want to be looking at actual voltages being used not what is being requested by the cpu

ive personally fixed the mb voltage by going into the parameters (double click the line or right click and choose parameters) and changing the ri kilo-ohm value to 0! 
I have left rf at 10 and left vf at 0, this now seems to read the exact same as cpu-z and hardware monitor :P 

not sure how to code so no clue how to implement this into the code and submit it towards the master but if anyone else can it would be greatly appreciated, otherwise i hope this helps anyone else noticing the same issue until its coded in :P

also huge thankyou to @PhyxionNL your doing amazing work keep it up!

STRETCHDEATH, 2024-08-14:
[image]
[image]

first screenshot if with rf set to 0 
second is set to default

Tellinq, 2024-08-23:
Mind also showing that second Nuvoton chip? I’m also wanting to see the rest of the sensors such as the temperature and fan sensors so I can get an idea of the proper data sheet I’d need to create.

I can probably add the first Nuvoton chip just fine, although I’m currently waiting on #1278 to get merged as that covers multiple SuperIO indexes

STRETCHDEATH, 2024-08-25:
[image]

oryxcallotis, 2025-02-14:
I noticed this too, with the B850 Pro RS WiFi.

[image]

Then a couple screenshots of the ri kilo-ohm value to 0;

[image]

[image]