LibreHardwareMonitor/LibreHardwareMonitor #1447: High temperature readings on Asus Prime B650 Plus

L0Lock, 2024-09-02:
I'm using [Rem0o/FanControl.Releases](https://github.com/Rem0o/FanControl.Releases) which uses LHM to know the hardware temperatures.

It detects 6 sensors off my Asus Prime B650 Plus motherboard, and two of which (number 3 and 6) have significantly high temperatures, often in the 80s ⁰C even at startup doing nothing. I though something was odd, so I ran [HWiNFO](https://www.hwinfo.com/) to compare (folded sections have no temperatures).:

[image]

I don't see any of the corresponding temperatures.

Tellinq, 2024-09-11:
Temperatures 3-6 are typically auxiliary sensors and have little meaning. I’d just ignore them.

If you want them removed outright, I can add a data entry for your specific board getting all the necessary details, but I’d need more data such as the voltages, which fan index corresponds in UEFI (`CHA_FAN#`/`CPU_FAN) from LHM in comparison to HWiNFO.

L0Lock, 2024-09-11:
My understanding of the subject falls pretty short. But here is a screenshot of the three apps, I hope this can help:

[image]

Tellinq, 2024-09-11:
Showing the default LHM sensors does help give some insight.

Since FanControl is basically using LHM as a backend, we won’t need that for this use case 

On HWiNFO64, can you enable voltages so we can see what is listed on the ASUS PRIME B650-PLUS sections? The other sections in HWiNFO can be collapsed

On LibreHardwareMonitor, can you go to View -> Show Hidden Sensors. I’m wanting to see what sensors are available.

Once that’s complete I’d like to see an updated side by side comparison as you were doing.

Tellinq, 2024-09-11:
That gives me a good chunk and guess of what fans are which.

Here's what I've done:

I've made a new build that should identify your specific board. I've filled in what I'm confident in as-is, but the majority of this is just filling in as many potential indexes as possible, whether they exist or not.

https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions/runs/10819089566

I recommend using the net472 package.

I'll still want verification in the UEFI on which fan corresponds to which just to make sure, but I'm confident I got the layout right.

When you do have time to get into the UEFI, I also want to see what available temperature sensors are reported in the UEFI. I'm seeing that for the most part, the CPU, CPU Package, Chipset, VRM, and/or Motherboard should get reported, but that always differs from board to board.

I've attached a screenshot of what I'm looking for to see:
[image]

Tellinq, 2024-09-19:
Wasn’t sure if you got my previous reply. Just needing a screenshot via the latest build so I can see what temperature sources are available and what I got right/wrong. Once that’s done I should be able to have your board added officially. See previous comment for the build.

L0Lock, 2024-09-19:
Hi, sorry for the delay, I got drowned in daily work. 

Here are a few screenshots:

[image]  
[image]  
[image]  
[image]  

And a screenshot from the net472 you linked above:

[image]

Hope i got it right. Thanks for your help :)

Tellinq, 2024-09-20:
Perfect! Board entry should be merged soon. Once that’s done, FanControl should also update LHM shortly after

PR #1453 (2024-09-20): Add ASUS Prime B650-Plus
Closes #1447 - added an entry for this specific board to only show known sensors and name accordingly. Auxiliary sensors are not shown.
closed 2024-09-20 by PhyxionNL