LibreHardwareMonitor/LibreHardwareMonitor #1458: Model i5-1135G7 CPU cannot display CPU temperature

NianBroken, 2024-09-14:
Before making this issue, I tried the following, but the program never displays the CPU temperature:

1. Exit the program completely and then reopen “`LibreHardwareMonitor.exe`”.
2. Run the “`LibreHardwareMonitor.exe`” program as administrator.
3. Restart my computer.

My computer system information is as follows:
Version: `Windows 11 Professional`
Version number: `23H2`
Operating system version: `22631.4108`

[image]

NianBroken, 2024-09-14:
@moellerm @PhyxionNL @pomianowski @Werelds @Phyxion @Tellinq
Here is my Report file: [LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/17000981/LibreHardwareMonitor.Report.txt)

sebastian-dev, 2024-09-14:
`Ring0

Status: Installing driver "D:\下载\LibreHardwareMonitor-net472\LibreHardwareMonitor.sys" failed and file exists
First Exception: StartService returned the error code: 000000B7.
Second Exception: StartService returned the error code: 000000B7.`
Maybe a folder name problem , try: use a folder name without special characters.
Possibly also a problem that the Defender or a Security Suite is blocking the driver.

NianBroken, 2024-09-14:
@sebastian-dev This time I re-downloaded the LibreHardwareMonitor program and ran it under the English path, unfortunately it still doesn't show my CPU temperature and I didn't turn on anything related to system security or system protection.

I have a habit of disabling or deleting all virus protection programs first thing I do when my computer arrives.

Here is my Report file: [LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/17001245/LibreHardwareMonitor.Report.txt)

[image]
[image]
[image]

sebastian-dev, 2024-09-14:
Look at the log report file. The driver cannot be installed. The information cannot be read without the driver.

NianBroken, 2024-09-14:
@sebastian-dev So what do I have to do now to get it to display the CPU temperature?

closed 2024-09-15 by PhyxionNL

NickJohnson868, 2024-09-27:
先退出联想管家，然后重启软件试试，我这样做成功了