LibreHardwareMonitor/LibreHardwareMonitor #1480: No any fan detected in X870 Hero

TMavica, 2024-09-30:
as title

Tellinq, 2024-09-30:
Mind sending a screenshot of what you *do* see? Please also enable Hidden Sensors

I expected given X870’s release there might be some layouts that need adjustments or additions

TMavica, 2024-10-01:
[image]
just empty, all sensor enabled

Tellinq, 2024-10-01:
I meant in LibreHardwareMonitor, not FanControl

TMavica, 2024-10-02:
> I meant in LibreHardwareMonitor, not FanControl

sorry here this, only gpu fan
[image]

Tellinq, 2024-10-02:
What about at the top? I’m mainly wanting to see what shows up for the motherboard. If nothing, I want to see what it’s identified as

TMavica, 2024-10-02:
> What about at the top? I’m mainly wanting to see what shows up for the motherboard. If nothing, I want to see what it’s identified as

sorry for late reply
[image]

Wyllio, 2024-10-11:
Also having the same issue but on the ASUS ROG STRIX X870-I GAMING WIFI motherboard. No sensors show up on Libre Hardware Monitor but they all show up on HWInfo64.

Wyllio, 2024-10-11:
[image]

roooots, 2024-10-13:
Please reply here if you're interested: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/1490#issue-2583718112

Thanks,
r.

styledpixel, 2024-11-12:
Same problem, no sensor information showing in LHM (hidden sensors enabled);

[image]

HWInfo shows the sensors though;

[image]

---

@Tellinq - do you need the NCT6701D data sheet from Nuvoton to implement the sensors or is there a way to add this motherboard in without? I've tried to edit the code myself but to be honest, I've got very little experience with C#

Woudart, 2024-11-17:
**FYI:**  
Support for the **Nuvoton NCT6701D** was added in the following versions:

- **HWiNFO:** v8.04 (June 12, 2024)  
- **AIDA64:** v7.35.7030 Beta (October 17, 2024)

Here are the relevant screenshots:

[image]  
[image]

TMavica, 2024-11-28:
Hero same as Gaming wifi NCT6701D, latest version still doesnt not show any fan sensor

damesene, 2024-12-02:
Based on this [X post](https://x.com/gussunoyoyo/status/1841867114933846068), I tried setting NCT6799D instead of NCT6701D for my Asus ProArt X870E-CREATOR WIFI motherboard, and it works perfectly, including in the Fan Control, after replacing the library.

roooots, 2024-12-02:
> Based on this [X post](https://x.com/gussunoyoyo/status/1841867114933846068), I tried to use NCT6799D instead of NCT6701D in a library for my Asus ProArt X870E-CREATOR WIFI motherboard, and it works perfectly, including in the Fan Control, after replacing the library.

Mind sharing what and how you did, for those who don't use X?

damesene, 2024-12-02:
The X post: "It's super I/O. I don't know the spec but it works as NCT6799D on NetBSD."

Originally I used the id of NCT6799D defined in the chip.cs and it worked fine but I found probably more correct id "0xD806" in another fork.

You can check the [code here](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1547/commits/0458d37ddf85a30de5cd6dd2b53bf344f2b830b9)

tijmenvn, 2024-12-11:
Same issue on ASUS X870P Motherboard, also uses Novuton NCT6701D.

All sensors / fans are detected and showing in HWiNFO, but LibreHardwareMonitor won't show.