LibreHardwareMonitor/LibreHardwareMonitor #1483: MSI MAG X670E fan control does not work

hardpoorcorn, 2024-10-05:
I havve an MSI MAG X670E motherboard with a Nuvoton 6687D SIO controller.  I am able to read RPM setting, however, when setting control values, none of the fans respond to values set in the software.  I have ensured that bios fan control is disabled, and that the fans are correctly setup in BIOS as PWM (which they are), yet any value applied in control yields no change in RPM.

[image]
[image]

Alcolawl, 2024-11-07:
Hey, I created a ticket the other day for a MSI X870 motherboard that uses the same I/O controller (Nuvoton NCT6687D). Feel free to reference it here, https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/1513. The more we draw attention to MSI boards that use this chip, the greater the chance it gets added to LHM.

I also started digging into the code behind the issue and made some headway, so hopefully that gets the ball rolling.

Alcolawl, 2024-12-21:
Hey @hardpoorcorn , just wanted to follow up. Pretty sure we've come up with a solution for this issue on several different motherboards. You can check out my [LHM fork](https://github.com/Alcolawl/LibreHardwareMonitor) if you'd like to see how. 

If you want me to add support for your board send a screenshot of how LHM looks on your machine and I can give it a shot. So far it's worked for a few other boards.