LibreHardwareMonitor/LibreHardwareMonitor #1488: Fans are not being detected in MSI MPG X870E CARBON WIFI

Nuctus, 2024-10-10:
Hello,

I am writing to notify that librehardwaremonitor is not detecting any fans in Mobo MSI MPG X870E CARBON WIFI

There is any way to bring compatibility with this?

Do you need anything from me in order to help for resolving this problem?

Thanks a lot.

Fran

roooots, 2024-10-13:
Please sign with https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/1490#issue-2583718112 if you're interested. 

Thanks,
r.

Strickens, 2024-10-24:
Having the same issue currently. Aorus Elite X870 Wifi 7

Razor1993-Real, 2024-11-19:
Yes, the Fan detection is not working properly on my MSI MPG X870E CARBON WIFI too, pls fix :)

demomanca, 2024-12-20:
Hi team, looks like you just got a new BIOS that says it fixes issues with HWMonitor. Could someone try this and see if it sorted for LHM as well?

Nuctus, 2024-12-20:
@demomanca  still not working. already tested

demomanca, 2024-12-20:
Damn. @Nuctus would you mind providing a File-> Save Report from your board? I'd like to see if the fix that being worked on for the X870 Tomahawk will work for these boards as well.
`

Alcolawl, 2024-12-21:
Hey @Nuctus try giving the latest release of my LHM fork a shot. I believe we have a fairly good solution working now for LHM and FanControl. 
https://github.com/Alcolawl/LibreHardwareMonitor

AwesomeAwesomeAwesome, 2024-12-22:
I tried your build @Alcolawl and now LHM properly detects all the motherboard fans. Too bad this did not translate to Fan Control sensor detection. And I made sure to fully uninstall MSI Center, flash BIOS with the latest update etc. - still no luck in Fan Control, alas.

[LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/18222063/LibreHardwareMonitor.Report.txt)
[image]
[image]

These are all bequiet fans from the latest bequiet dark base pro 901 case.

Alcolawl, 2024-12-22:
> I tried your build @Alcolawl and now LHM properly detects all the motherboard fans. Too bad this did not translate to Fan Control sensor detection. And I made sure to fully uninstall MSI Center, flash BIOS with the latest update etc. - still no luck in Fan Control, alas.
> 
> [LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/18222063/LibreHardwareMonitor.Report.txt) <img alt="Screenshot 2024-12-22 145534" width="1309" src="https://private-user-images.githubusercontent.com/47180044/397990230-fe171ad8-7be9-4009-870f-9c098fd3e3ac.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ4OTE1NTAsIm5iZiI6MTczNDg5MTI1MCwicGF0aCI6Ii80NzE4MDA0NC8zOTc5OTAyMzAtZmUxNzFhZDgtN2JlOS00MDA5LTg3MGYtOWMwOThmZDNlM2FjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjIyVDE4MTQxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRjMzNjZDA0NDM2MTZiZmI5YTU4ZThiMTQyNGNjMWIzODM2MGQ2NTA2YzAzOGE3Mzc2MjU0Mzc5MDk4OGQ0YzYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.7Y7ialdEU8dyPafyupj1iA8bkD7EWChu4Un1IMZfT1M"> <img alt="Screenshot 2024-12-22 140153" width="1431" src="https://private-user-images.githubusercontent.com/47180044/397990234-3ce6bc64-0fa9-4d56-8e4a-4aaea2b571cd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ4OTE1NTAsIm5iZiI6MTczNDg5MTI1MCwicGF0aCI6Ii80NzE4MDA0NC8zOTc5OTAyMzQtM2NlNmJjNjQtMGZhOS00ZDU2LThlNGEtNGFhZWEyYjU3MWNkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjIyVDE4MTQxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZhMmYzNjc4NmMwZjc4NzUzNjE1ODEyZmRjYzFhNzI3NzAzMzdhODUxYThmMzVlOTFiZTFhMGJkNjUwODM2NDEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.QBzSkXoN1eu4r-Jk9G8STbgLSADKcjC4FOCdhZyrzD0">
> 
> These are all bequiet fans from the latest bequiet dark base pro 901 case.

Pair the speed sensor to the fan and you'll be all set, just click the "Pair speed sensor" UI element.

I haven't quite figured this bit out yet, it might be a FanControl limitation. But basically FanControl _incorrectly_ thinks the speed sensor is incorrect because the fan is slow to respond to the speed changes. For whatever reason, the fans, when commanded, are slower to ramp up and down than expected. So when you set a fan to 100%, it takes longer than the program expects to spin up to full speed. The program basically runs out of patience (some kind of interval is exceeded) and thinks none of the sensors are correctly matched up with the fan, so you have to force it yourself. If you do an auto calibration on the fan speed sensors you'll see what I mean.

iliketoprogram14, 2025-01-13:
Do you need any more info or logs to get this working?

Alcolawl, 2025-01-13:
> Do you need any more info or logs to get this working?

Appears to be working fine for several X870E Carbon owners.

gionkunz, 2025-01-18:
I was finally able to ditch MSI Catastrophy Center!!! Thanks @Alcolawl for the fork and your work!

[image]

FreakyPriest, 2025-01-28:
The fork is working for the MSI MPG X870E CARBON. How are the chances that this ends up in the official release?

demomanca, 2025-01-28:
We're discussing that at the moment. Given this is a novel way for LHM to control fans, the maintainers might not want it in. We're pushing a change to enable what we can control in the existing way LHM does it, and with that I'd say we'll discuss this method.