LibreHardwareMonitor/LibreHardwareMonitor #1490: How may I/we assist with implementing X870E sensor support?

roooots, 2024-10-13:
Good morning all,

the title says it - having happily used LHM/FanControl for several years, with the new AMD X870E chipset sensor support is obviously lacking.

I'd be happy if I could assist with implementing X870E sensor support and although my coding skills are rather rudimentary, I could do research, gather data and run tests for my particular mainboard at least. As a Linux guy, I'm comfy with the command line.

Anyone supporting this idea please reply with their precise mainboard and OS type and how they could assist.

--

Mainboard: Asus Strix X870E-E Gaming Wifi, latest 0505 BIOS.
OS: Windows 11 24H2
Skills: Basic coding, data gathering and analysis, testing

--

Cheers,
r.

roooots, 2024-10-13:
Btw. - here's a screenshot taken with LHM latest nighty, this is with only mainboard and fan categories enabled.
[image]

Strickens, 2024-10-24:
Posting here to bump as well. Sensors are detected in HWMonitor but not Libre Hardware Monitor.
[image]

DavidMollmann, 2024-10-31:
Have a look at [this PR](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1510) if you're interested. This might solve your Problem.

oxivanisher, 2025-01-16:
I have the same mainboard and I don't think this is solved yet. Here are my infos

Motherboard: ASUS ROG STRIX X870E-E GAMING WIFI
Bios version: 0804
Fans connected: CPU, AIOPUMP (as SYS), AUX 3, AUX 4
Chip: Novuton NCT6799
[HWMonitor.txt](https://github.com/user-attachments/files/18435404/HWMonitor.txt)
[LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/18435405/LibreHardwareMonitor.Report.txt)