LibreHardwareMonitor/LibreHardwareMonitor #1493: Bad CPU core clock reporting on AMD Granite Ridge 9000 series

jerr060599, 2024-10-15:
I built the library with the latest source code for 23207acb098490523ca1b68488314a88489c4a58. 
On the 9950x, I'm getting bad sensor values for core frequency.
 
Randomly, the first three or so cores of CCD1 would suddenly report having a clock of 14582.668
There is simply no way that is true.

MasterLink86, 2024-10-23:
I personally have been observing random issues with the wrong CPU core clocks being reported for around 5+ years now (going back to OpenHardwareMonitor) with Core 2 Quads, AMD FX, and now my Ryzen 5 8000 series.  It seems this bug has been quite persistent, because today it claimed my Ryzen 5 was idling at 7GHz.  That's simply not happening.  It only seems to happen though most when it starts with logon (when it's launching while other apps are launching).  I have found if you launch it with nothing else actually consuming resources, the chances of accurate clocks are more likely, but still never guaranteed.

I'd like to see this ultimately resolved, but after 5+ years, and observing the issue with OpenHardwareMonitor as well, I'm not holding hope at this point as I had to opt for other reporting utilities, but nothing ever is as comprehensive as LibreHardwareMonitor with the desktop gadget/widget, which is a shame.

LFAB3, 2024-11-20:
Hello,
Exist a new issue cloned. Sorry for comment this in new one instead this. 
I am attaching an image where it can be observed that the frequency shown far exceeds 10,000 MHz compared to CPU-Z.
[image]

sebastian-dev, 2024-12-03:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1548