LibreHardwareMonitor/LibreHardwareMonitor #1513: No fans detected on MSI X870 Tomahawk WiFi (Nuvoton NCT6687D)

Alcolawl, 2024-11-03:
Title says it all. Running the latest version of LibreHardwareMonitor. No fans, other than the CPU fan, are detected.

Motherboard: MSI X870 Tomahawk WiFi

Attached is a screenshot of LHM with Hidden Sensors enabled. 
[image]

As you can see, I think the **Nuvoton NCT6687D** is the I/O chip we want to be targeting here. Please let me know if you need any other information. I suspect other 800 series MSI motherboards may use the same or similar sensor.

Alcolawl, 2024-11-03:
Referencing #1490 for more visibility. As stated earlier, there's a fair chance other MSI 800 series boards use the Nuvoton NCT6687D.

perlakiz, 2024-11-04:
Hi, I am experiencing the same issue with my Z890 motherboard, which also uses the Nuvoton NCT6687D controller. #1516

Alcolawl, 2024-11-06:
HWInfo seems to detect the fans just fine (Although far too many of them... Duplicates?)
[image]
System Fan 1 and 2 are, in-fact, in use in my system for the front and rear fans respectively. So what HWInfo is reporting is correct. You'll notice that the control for System Fan #2 reports in properly but changing the value manually has no effect. So:

- LHM is not detecting all of the fan headers populated in the system
- The detected Controls have no effect

(_Note: Some of the voltage names have changed since my last screenshot since I'm working the problem from my end as well and the voltages reported in the current release are mislabeled._)

I'm beginning to hit a wall here, since I'm not entirely too familiar with the code. I just decided to tear it apart since I write code for my day job and I wanted to help.

Alcolawl, 2024-11-06:
After some trial and error in 
`LibreHardwareMonitor/blob/master/LibreHardwareMonitorLib/Hardware/Motherboard/Lpc/Chip.cs`

I discovered where System Fan 1 and 2 are hiding. I had to add 8 new Fan RPM Registers. 
[image]

The code that produces the RPM results is as follows:
```
           case Chip.NCT6687D:
                Fans = new float?[16];
                Controls = new float?[8];
                Voltages = new float?[14];
                Temperatures = new float?[7];
                
                # Non-Pertinent code removed for brevity
                
                // CPU Fan 0x140
                // PUMP Fan
                // SYS Fan 1 0x160
                // SYS Fan 2 0x15E
                // SYS Fan 3 0x15A
                // SYS Fan 4 0x158
                // SYS Fan 5 0x156
                // SYS Fan 6 0x154
                _fanRpmRegister = new ushort[] { 0x140, 0x142, 0x144, 0x146, 0x148, 0x14A, 0x14C, 0x14E, 0x150, 0x152, 0x154, 0x156, 0x158, 0x15A, 0x15C, 0x15E };
```

I have a feeling the array of registers are stored and iterated through backwards. You'll note that I commented the HEX value of the registers I've identified.

Not sure how to handle this, other than writing a special case for this particular motherboard, because a lot of this NCT6687D code was written by someone else and presumably worked fine for the majority of the boards that used this chip until now. Now to get the controls working...

Tellinq, 2024-11-06:
For the controls you’ll likely need to mirror any changes that you do for the fans (instead of 8, specify 16 like you did with the fans) - it does mean you’ll likely need to identify each fan register so the controls then work

Alcolawl, 2024-11-06:
> For the controls you’ll likely need to mirror any changes that you do for the fans (instead of 8, specify 16 like you did with the fans) - it does mean you’ll likely need to identify each fan register so the controls then work

@Tellinq 
I tried that. I expanded the array to a length of 16 and added like 8 new test controls but none of them populated with values and none of them actually controlled anything.

I shortened the register array back to 8 and excluded some of the registers that didn't seem to align with an actual fan headers on the motherboard (These can be recognized by the vague "Fan 4,5,6, etc in HWInfo). I also put the registers in reverse order so they'd appear in the proper order in LHM. 

```
Fans = new float?[8];
...
//_fanRpmRegister = new ushort[] { 0x140, 0x142, 0x154, 0x156, 0x158, 0x15A, 0x15C, 0x15E };
_fanRpmRegister = new ushort[] { 0x140, 0x142, 0x15E, 0x15C, 0x15A, 0x158, 0x156, 0x154 };
```
[image]

So I think I've identified the correct registers (At least for CPU Fan, System 1 and 2), but only the CPU Fan control works. I can feed the System fan controls values from FanControl or just manually through LHM but they have no effect. I have a feeling that this is due to the controls not being detected or set properly but I haven't been able to figure it out. Would they too also be detected in reverse order like the RPM registers?

Alcolawl, 2024-11-06:
Through trial and error, I've discovered a register than gets the PWM Signal % readout for System Fan 1. Look at the comments and code below:
```
            /*REGISTER KEY:
            FAN_PWM_OUT_REG = Register that stores the value of Fan Control in % (0-100)
                0x151, 0x153, 0x155, 0x157 = NOTHING? Reports 0
                0x159, 0x15B = Just counts backwards?
                0x15D = Useless
                0x15F = System Fan #1
                0x160 = CPU Fan
                0x161 = "Pump Fan" could be wrong
                System Fan #1, despite being populated, does not report in
                System Fan #2, despite being populated, does not report in
                System Fan #3, reports a value but doesn't coincide with anything.
                System Fan #4, despite being populated, does not report in

            FAN_PWM_COMMAND_REG = Register that manipulates the fan
                0xA28 = CPU Fan
                0xA29 = "Pump Fan" could be wrong                
            */
             FAN_PWM_OUT_REG = new ushort[] { 0x160, 0x161, 0x15F, 0x162, 0x163, 0x164, 0x165, 0x166 }; // Effects Control Readout
             FAN_PWM_COMMAND_REG = new ushort[] { 0xA28, 0xA29, 0xA2A, 0xA2B, 0xA2C, 0xA2D, 0xA2E, 0xA2F }; // Manipulates the fan
```
System Fan 1's PWM Signal % value is stored in register 0x15F. Since 0x160 is the CPU fan, and 0x161 is the Pump Fan. This would lead you to believe that System Fan 2's register would be 0x15E, Sys_Fan 3 = 0x15D, so on and so forth. However, half the registers from numbers 0x150 to 0x15F are dedicated to storing Fan RPMs:
```
                // CPU Fan 0x140
                // PUMP Fan 0x142
                // SYS Fan 1 0x15E
                // SYS Fan 2 0x15C
                // SYS Fan 3 0x15A
                // SYS Fan 4 0x158
                // SYS Fan 5 0x156
                // SYS Fan 6 0x154
                _fanRpmRegister = new ushort[] { 0x140, 0x142, 0x15E, 0x15C, 0x15A, 0x158, 0x156, 0x154 };
```
I've confirmed this by plugging a fan into every slot on my motherboard and confirming the readout in LHM. Those registers are correct. You'll notice that we skip all odd numbered registers in that range, so I tried plugging those in to see if they were the PWM Signal % values but they're not. They either readout 0% or count down backwards. **They are not PWM Signal % registers.** 

So this confuses me a bit. Whoever originally wrote this code expected registers 0x160 - 0x167 to store fan control values, but only 2 of them appear to do so. Furthermore, 0x15F, a lower number outside of the expected range, belongs to System Fan 1. I have no idea where to go from here in terms of registers for PWM Signal readout.

MysfitLeaf, 2024-11-10:
Hi, I have the exact same motherboard as you. Have you made any headway in figuring out the issue past what you have already done. I dont know much about this kind of thing but if you need anything data ect.. may be able to provide.

Thanks!

Alcolawl, 2024-11-11:
@MysfitLeaf Unfortunately, I haven't. I previously resorted to basically guessing because there's no datasheet or reference to give me guidance on where to look next. I just don't have the time in the day to waste guessing between work and real life stuff. I reached out to Nuvoton for a NCT6687D-R datasheet and they got back to me last night. They referred me to MSI, despite not having mentioned them in my email, so it looks like it's a bespoke part for that vendor and, for whatever reason, they have no supporting documentation on it. I'm going to email MSI shortly, but I'm not expecting to get much information from them.

Alcolawl, 2024-11-11:
@MysfitLeaf MSI responded with the following:

> "We would not be able to provide such documentation and we unsure why the manufacturer of the module would tell you this. We do not send out internal documentation."

I'll go back to Nuvoton with this information and see what they say but this looks like a dead end. Ridiculous. 

And just to chime in on what you said:
> Annoying that its not just supported straight up.

The code is there that _should_ support it but the NCT6687D-R appears to be just a tad bit different than the NCT6686D that's been used by other boards in the past. Looks like a lot of X870 boards use this chip this time around, so the community needs to come together to figure this out.

Alcolawl, 2024-11-11:
This may be helpful:
https://github.com/Fred78290/nct6687d/blob/main/nct6687.c

I'm not too familiar with C, the language, but there may be some key pieces of information in here on how to modify the LHM code to better support the NCT6687D.

MysfitLeaf, 2024-11-24:
[image]

Had a look at another fan control software (Argus Monitor), and it seems to be able to detect RPM of the fans, but shows them as "Not Controllable" Has two lists both "Fan Control" one with where you would expect the fans to be and missing all RPM detection except from the CPU and AIO header on the motherboard. It is possible that a workaround for now to try and put all the fans on the CPU header or the AIO header so that they can be controlled. Something only advisable if you have a fan controller that is powered seperately so the load on the fan port is not too heavy.

Got no clue how to go about incorporating this into a fix for LHM. but thought it might help ?

brianpardee, 2024-12-11:
Hello, I'm wondering please if there has been any updates on this from anyone with the tomahawk board? I attempted to copy and paste and fill in values as best I could from another pull request for a different x870 board. I was able to see the speeds in fancontrol using my build, but I wasn't able to control them.

Alcolawl, 2024-12-12:
> Hello, I'm wondering please if there has been any updates on this from anyone with the tomahawk board? I attempted to copy and paste and fill in values as best I could from another pull request for [...]

Unfortunately I've basically given up. I'm not sure how developers identify registers, probably with some kind of tool or software to tap into the I/O controller or a certain bus, and I don't have one, nor do I plan on purchasing one. I almost get the feeling that this is an X870 related issue. Other earlier motherboards that use the Nuvoton NCT6687D have zero issues, and I've looked around quite a bit. It only seems to be related to 800 series AM5 boards for whatever reason.

I've reached out to Nuvoton, who basically refuses to help, I've poured over their NCT6686 Family documentation, etc. I've run out of resources. The code, as it stands, should work, and does on older motherboards with the same exact Super I/O chip, and yet it doesn't for newer boards. Fan RPM registers buried where they shouldn't be, being enumerated in reverse order. Something just isn't making sense here. Even [this guy's NCT6687D code](https://github.com/Fred78290/nct6687d/issues/107), of which was actually merged into various Linux distro kernels, doesn't seem to work with X870 boards. I'm not sure what else to do at this point. Everyone thus far has either missed something extremely simple, or there's been an actual change with the AM5 chipset or Nuvoton chip that people are unaware of that is causing all of this.

Dr-Mugg, 2024-12-12:
> Unfortunately I've basically given up. I'm not sure how developers identify registers, probably with some kind of tool or software to tap into the I/O controller or a certain bus, and I don't have [...]

I have a MSI MPG Carbon Z890 with the same chip and problem so i don't think its related to just X870.

Alcolawl, 2024-12-12:
> I have a MSI MPG Carbon Z890 with the same chip and problem so i don't think its related to just X870.

Good point. Perhaps it's a change in the Nuvoton chips themselves.

Zabojnikk, 2024-12-15:
> ![Screenshot 2024-11-24 004900](https://private-user-images.githubusercontent.com/187995911/389239129-b9f3859c-1aec-4770-bd70-4b69440a8de2.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQyMjUzMzMsIm5iZiI6MTczNDIyNTAzMywicGF0aCI6Ii8xODc5OTU5MTEvMzg5MjM5MTI5LWI5ZjM4NTljLTFhZWMtNDc3MC1iZDcwLTRiNjk0NDBhOGRlMi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxNVQwMTEwMzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yNWViZWQyYTA2MjJhZmVjNzdiZWIwYWNiNjhjMzMwMGJlNTA3MjQwZjhjM2Y0M2MwNmE5M2Y0NjFjYWExNmZiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.VeSn0c0DyW754qR4SwjEkMcn8w43uYLO639CZPKup5o) ![Screenshot 2024-11-24 010206](https://private-user-images.githubusercontent.com/187995911/389239441-3c8873ee-68ce-442f-bd89-fa577d4405d8.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQyMjUzMzMsIm5iZiI6MTczNDIyNTAzMywicGF0aCI6Ii8xODc5OTU5MTEvMzg5MjM5NDQxLTNjODg3M2VlLTY4Y2UtNDQyZi1iZDg5LWZhNTc3ZDQ0MDVkOC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxNVQwMTEwMzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iNTM4NTMzYjBiZGVmNGFhZmNmYzFmMjZjM2FhZTE1YzU3NjMyZmU4YTU4ZGY0OWMzMzk3YWQxY2FlNzk0YWRlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.EQ00cyf1hHGfy2uZumvgq5Ilm0ZlDWgbepMZNio_XhU)
> 
> ![Screenshot 2024-11-24 010151](https://private-user-images.githubusercontent.com/187995911/389239442-ca938dc5-997a-4928-a5c5-aa4ec04a1ed4.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQyMjUzMzMsIm5iZiI6MTczNDIyNTAzMywicGF0aCI6Ii8xODc5OTU5MTEvMzg5MjM5NDQyLWNhOTM4ZGM1LTk5N2EtNDkyOC1hNWM1LWFhNGVjMDRhMWVkNC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxNVQwMTEwMzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lYjBhNzc5YzRhNjVlZTJkMzJkMGUxZjE5NGFkYjk1NDE1YzIxN2QzYmFjYzE3NWEyNmYxYzU1NDMzZGZlZmJkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.QFott1os_qOmdLWHNsRrEZsyhtRUPXceknEQkgQ0Ijs)
> 
> Had a look at another fan control software (Argus Monitor), and it seems to be able to detect RPM of the fans, but shows them as "Not Controllable" Has two lists both "Fan Control" one with where you would expect the fans to be and missing all RPM detection except from the CPU and AIO header on the motherboard. It is possible that a workaround for now to try and put all the fans on the CPU header or the AIO header so that they can be controlled. Something only advisable if you have a fan controller that is powered seperately so the load on the fan port is not too heavy.
> 
> Got no clue how to go about incorporating this into a fix for LHM. but thought it might help ?

I don't know if there's been an update to Argus Monitor that fixed the above issue, but it's detecting & reading all fan headers correctly for me, all perfectly controllable.

I wanted to give Fan Control a proper go with my new PC, and I might still play with the HWInfo plugin, but, ehhh. Might just stick with Argus, which has served me brilliantly over the past few years, it just don't have the extra fine control FC does.

demomanca, 2024-12-17:
Just picked up one of these (X870 Tomahawk) and got inspired to find the damn registers because I want this and FanControl to work as well.

Working from what @Alcolawl found:
Fan RPM registers are indeed 0x140, 0x142, 0x15E, 0x15C, 0x15A, 0x158, 0x156, 0x154
I believe the speed sensors are 0x160, 0x161, 0xE05, 0xE04, 0xE03, 0xE02, 0xE01, 0xE00
and I _think_ the controls are at 0xA28, 0xA29, 0xC9F, 0xC9D, 0xC9B, 0xC99, 0xC97

It's very late here in Aus, but with those changes made to line 315, 74 and 75 respectivly in Nct677x.cs I'm able to read fan RPMs and PWM speed % settings, but only CPU is able to be controlled.

Has this got to do with line 76 and 77 all repeating maybe?

demomanca, 2024-12-17:
> The Speed Sensor registers appear to be correct. May I ask how discovered these?

I went into the bios and set each of the sys fans to fixed 90%, 70%, 50%, 30% and 0%, then looked for that pattern in the register dump from LHM. 

I'm thinking something similar might be able to be done if I use the MSI Fan speed controller app (yuck) to control the fans at similar levels, and then pull the registers again.

Alcolawl, 2024-12-17:
> I'm thinking something similar might be able to be done if I use the MSI Fan speed controller app (yuck) to control the fans at similar levels, and then pull the registers again.

Oh, very nice. I tried saving a report previously but I guess I scrolled right past the register dump. That certainly makes things a bit easier! I already have the crappy MSI software installed since my current BIOS version broke the in-BIOS fan control, so I went ahead and gave it a shot.
[image]
I ran a report before and after I manually set my Two system Fans to 100%. There were a few difference in the register dump, as one would expect when temperatures and voltages and fluctuating, so these registers are really the only ones that make sense; especially ones set to full value of FF (255). There's quite a few FF values to choose from but I'd say our answer lies in there.

demomanca, 2024-12-17:
[Register Analysis.xlsx](https://github.com/user-attachments/files/18172266/Register.Analysis.xlsx)

Another set of eyes won't hurt.

I created an excel sheet with 3 register dumps, 1st is my normal config, which is CPU fan moving with cpu temp, pump @ 100%, then sysfan 1-3 all moving with system temp, called Auto, 2nd is using LHM to manually set CPU fan to 100% (which works, and points at 0x160), everything else the same, and 3rd is my manual "spread" of fan speeds that I used to detect the registers. I then got it to highlight changes, and I also translated to %s underneath. Maybe you can spot something I've missed?

demomanca, 2024-12-17:
Just playing with CPU fan control, I actually don't think it's working correctly. I can manually set it, and it goes to those speeds, but when you put it back to default something funky is happening. I've noticed if you set the speed manually to above 50%, when you go back to default, the fan returns locked to 50%, and won't move with cpu temps again, but if you set it lower than 50%, then put it back to default, it returns to normal smart operation. Must be some register that's not being set correctly on the way back to default.

Playing with MSI Centre to set some speeds and read back, 0xC40 through 0xC70 are moving around, I'll start there.

Alcolawl, 2024-12-17:
> [Register Analysis.xlsx](https://github.com/user-attachments/files/18172266/Register.Analysis.xlsx)
>
> Another set of eyes won't hurt.
>
> I created an excel sheet with 3 register dumps, 1st is my normal [...]

Nothing particularly stands out to me except two lines. First the easy one. 

The current code has FAN_CONTROL_MODE_REG set to an array 8 elements all referencing the value stored in register 0xA00. Looking at your excel sheet comparing it to my own register dumps ours are completely different, yet we both still have the ability to control the CPU fan. That's interesting.

But the set of registers that really stood out to me was 0B00. In comparing my three register dumps: 

1. Fans on auto and not being set to a strict value, 
2. All of the fans in my system set to 100% (I only have 3 headers populated), 
3. Fans back on auto, 

I discovered found the first 7 registers, 0B00 - 0B06 changed from `00 2D 41 50 50 50 50` --> `20 20 20 20 20 20 20`. Furthermore, registers 0B08-0B0C changed from `33 00 59 00 B2`  --> `FF 00 FF 00 FF`. The results on your excel sheet confirm this to a degree, with the same set of registers in your 2nd Reg dump reflecting the values I had when my fans were set on auto, and your 3rd reg dump reflecting the values I had when my fans were set to 100%.

Not sure what it means just yet but I figured I'd share. I have a feeling that much like other Nuvoton chips, we might have to reference different registers to control each fan header individually. [Example](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitorLib/Hardware/Motherboard/Lpc/Nct677X.cs#L88)

demomanca, 2024-12-18:
Ok, for speeds, I found the following:
Sys 1: 0C70, 0C72, 0C74, 0C76, 0C78, 0C7A, 0C7C
Sys 2 (all 24 register down) 0C58, 0C5A, 0C5C, 0C5E, 0C60, 0C62, 0C64
Sys 3 (again, 24 down) 0C40, 0C42, 0C44, 0C46, 0C48, 0C4A, 0C4C

Also found 0C9F, 0C9D and 0C9B for Sys 1, 2 and 3 (as well as the previously discovered 0E05-0E03 run)

Need to do paid work for a bit, maybe we attack these with control and see what bites?

demomanca, 2024-12-18:
So I set:
`            FAN_PWM_OUT_REG = new ushort[] { 0x160, 0x161, 0xE05, 0xE04, 0xE03, 0xE02, 0xE01, 0xE00 };`
`            FAN_PWM_COMMAND_REG = new ushort[] { 0xA28, 0xA29, 0xC70, 0xC58, 0xC40, 0xC28, 0xC10 };`

This started doing strange stuff, but was impacting the fan speeds. Setting to 100% on any fan set all Sys fans to 50%.

So, this is my first foray into any sort of programming for more than 30 years tbh, so bear with me. Using a combination of my own brain and ChatGPT, I think that we're still either not quite on the PWM_COMMAND_REG numbers, or, because they're in a different area to the CPU and Pump fans, they need a different way to write to them (across 16 bits maybe?). On top of that, again, because they're in different registers to the CPU and Pump, I'm assuming 0xA00 isn't the correct FAN_CONTROL_MODE_REG i for these 6 sys fans either. I am however reasonable confident in:

`FAN_PWM_OUT_REG = new ushort[] { 0x160, 0x161, 0xE05, 0xE04, 0xE03, 0xE02, 0xE01, 0xE00 };`

No matter what wackiness happens with the fans, these are always reading the right speed against the RPM of the fans.

The search continues. Tell you what, some reference docs wouldn't go astray right about now.

Alcolawl, 2024-12-18:
> So I set: ` FAN_PWM_OUT_REG = new ushort[] { 0x160, 0x161, 0xE05, 0xE04, 0xE03, 0xE02, 0xE01, 0xE00 };` ` FAN_PWM_COMMAND_REG = new ushort[] { 0xA28, 0xA29, 0xC70, 0xC58, 0xC40, 0xC28, 0xC10 };` [...]

I hear ya. I'm kind of at my wit's end here too. I program OS and Hypervisor automation for a living, have a background in C++, but this is my first time playing around with C# and definitely my first time attempting to interact with a Super I/O controller, so we're in the same boat. I agree that documentation would go a long way here. As mentioned much earlier in this Issue thread, I reached out to Nuvoton, who strangely told me to contact MSI. When I replied to Nuvoton via email, I received no response. Perhaps it's because I didn't respond in their customer service portal, but when attempting to open the supplied link, it leads to a completely broken and unusable webpage; I tried it in several browsers. The rep's claim was that they don't have the datasheet for their own product, which is absolutely false. I'm not sure what the big secret is. Perhaps you or someone else following along can try their luck reaching out to them. 

I think we're in a bit of a rut. I spent entirely too long yesterday staring at this stuff, and it looks like you have too. On top of the fact that we're on opposite sides of the planet and have actual jobs, correspondence will have huge lag times. I think maybe we reset with a map of current known registers, it can't hurt, and this way it identifies what we know, what we don't, and what registers we don't need to bother looking at more and can begin to look elsewhere. A major blocker here too is that, while the register dumps are nice, they're a snapshot in time. Having a utility that could constantly give us a readout of all the registers would be very helpful as I suspect quite a few of them are counting up or down, iterating, changing, flipping, etc. There are some interesting patterns of register values in there that seem to alternate without rhyme or reason.

Here's what we have so far:

```
Temperature Registers = 0x100 - 0x10C (Even numbered registers only)
Voltage Registers = 0x120 - 0x13C (Even numbered registers only)

Fan RPM Reporting Registers: 0x140, 0x142, 0x154 - 0x15E (Even numbered registers only)
CPU Fan 0x140
PUMP Fan 0x142
SYS Fan 1 0x15E
SYS Fan 2 0x15C
SYS Fan 3 0x15A
SYS Fan 4 0x158
SYS Fan 5 0x156
SYS Fan 6 0x154
EZ Connect ???

Fan PWM Signal (%) Registers: 0x160, 0x161, 0xE00 - 0xE05
CPU Fan 0x160
PUMP Fan 0x161
SYS Fan 1 0xE05
SYS Fan 2 0xE04
SYS Fan 3 0xE03
SYS Fan 4 0xE02
SYS Fan 5 0xE01
SYS Fan 6 0xE00
EZ Connect ???

Fan Control (%) Registers: 0xA28, 0xA29, 0xC97 - 0xC9F (Odd numbered registers only)
CPU Fan 0xA28
PUMP Fan 0xA29
SYS Fan 1 0xC9F
SYS Fan 2 0xC9D
SYS Fan 3 0xC9B
SYS Fan 4 0xC99
SYS Fan 5 0xC97
SYS Fan 6 ??? (We're missing a register here, only 7 values instead of 8.)
EZ Connect ???
```
I've mapped out and color coded these values, as well as some others (by reading through the code) with this [Google Sheet](https://docs.google.com/spreadsheets/d/1kWWF88rG7b-muFXRmnHkFApQDs0LpDC02vnP9YEZHQ0/edit?usp=sharing). Feel free to add your own discoveries. I've also saved a backup in case anyone tries to get cute and vandalize our work.

**Question is:** why would Nuvoton change up all the registers on the NCT6687D out of the blue without changing the model number? Doubt that's common practice. Maybe that's what the "-R" designation is. Something's just not adding up, here.

demomanca, 2024-12-18:
I had a read of https://www.nuvoton.com/resource-files/NCT6686D_HW_Datasheet_V0_5.pdf and it seems that chip (so maybe this chip too) has some "flexible" controls that can be used for whatever, and maybe MSI's using one of these for the sys fans, which is why they're in completely different registers.

I'm about to unplug for a few weeks for the holidays and I'll be away from my main PC (and this chip) for a bit. I'll keep an eye on any other PCs I walk past for this chip (without seeming weird asking strangers what SuperIO chip they have :) )

ppittle, 2024-12-19:
> without seeming weird asking strangers what SuperIO chip they have

nothing weird about that!  

Really hope you guys can crack this.  I just built a new system using a X870 MSI Tomahawk and was looking forward to setting up [FanControl](https://github.com/Rem0o/FanControl.Releases).

demomanca, 2024-12-19:
Ok, here's what I think I've found:
I think

 - CPU and Pump Controls work, but have a bug, it takes 2 settings of default to make them go back to default. The first time to control it, and set it back to default, it gets stuck at 50%, then taken control again, and set it back again, and it goes back to bios mode.
 - MSI's software sets the speeds manually by manipulating ALL the points on the speed control curve to the speed it wants. The speed control curves are located at:
	
	 - Sys 1 > 0xC7C to 0xC70
	 - Sys 2 > 0xC64 to 0xC58
	 - Sys 3 > 0xC4C to 0xC40
	 - Sys 4 > 0xC34 to 0xC28
	 - Sys 5 > 0xC1C to 0xC10
	 - Sys 6 > 0xC04 to 0xBF8
Each Skips a register
 - Something is definatly happening at the 0xB0x and 0xB1x registers as well. I think it might be the curves for the CPU and Pump headers, same as the sys ones I've listed above. Again, MSI's software pushes all these to what speed you set, but LHM doesn't. 

There must be another set of registers similar to 0xA28 and 0xA29 that LHM can use to set the speed rather than making all the curve points the speed, but given the MSI software doesn't move 0xA28 and 0xA29 when it sets it's speed, I don't know that I can find it by moving fan speeds around.

This also explains why when I used the bottom registers (0xC70, 0xC58 etc) it was doing something. It was pushing the low end of the fan curve up, which would change the speed.

Lastly, I found some LHM dumps from other NCT6687D boards that aren't x870 or z890 from some of the other comments/pull requests here. It seems on those boards, 0xD00 and up are basically empty/not used, as they read all 00, but there is curve info in the registers listed above.

I've attached lots of examples from my system, where I've set the fans to variations of smart and fixed.

[MSI Bios Mode.txt](https://github.com/user-attachments/files/18202551/MSI.Bios.Mode.txt)
[MSI Even Smart.txt](https://github.com/user-attachments/files/18202552/MSI.Even.Smart.txt)
[MSI Odd Smart.txt](https://github.com/user-attachments/files/18202553/MSI.Odd.Smart.txt)
[MSI Stepped v2.txt](https://github.com/user-attachments/files/18202554/MSI.Stepped.v2.txt)
[MSI Stepped v3.txt](https://github.com/user-attachments/files/18202555/MSI.Stepped.v3.txt)
[MSI Stepped.txt](https://github.com/user-attachments/files/18202557/MSI.Stepped.txt)
[Non x870 example.txt](https://github.com/user-attachments/files/18202558/Non.x870.example.txt)
[MSI all set to 100%.txt](https://github.com/user-attachments/files/18202559/MSI.all.set.to.100.txt)
[MSI all smart (another example).txt](https://github.com/user-attachments/files/18202560/MSI.all.smart.another.example.txt)
[MSI all smart.txt](https://github.com/user-attachments/files/18202561/MSI.all.smart.txt)

Alcolawl, 2024-12-19:
> Ok, here's what I think I've found: I think
>     * MSI's software sets the speeds manually by manipulating ALL the points on the speed control curve to the speed it wants. The speed control curves are located at:
>       
>       * Sys 1 > 0xC7C to 0xC70
>       * Sys 2 > 0xC64 to 0xC58
>       * Sys 3 > 0xC4C to 0xC40
>       * Sys 4 > 0xC34 to 0xC28
>       * Sys 5 > 0xC1C to 0xC10
>       * Sys 6 > 0xC04 to 0xBF8
>         Each Skips a register
> 
> There must be another set of registers similar to 0xA28 and 0xA29 that LHM can use to set the speed rather than making all the curve points the speed, but given the MSI software doesn't move 0xA28 and 0xA29 when it sets it's speed, I don't know that I can find it by moving fan speeds around.
> 
> This also explains why when I used the bottom registers (0xC70, 0xC58 etc) it was doing something. It was pushing the low end of the fan curve up, which would change the speed.

Amazing work on the fan curve registers, I actually just checked this on my own system and it's correct. I can't believe I didn't catch this earlier. 

Example, I have the following curve setup in MSI Center for System Fan 1:
[image]
Steps in text as as follows:
```
PWM % | Temp (°C)
40 | 40
52 | 45
75 | 48
100 | 50
```
So I pulled the registers for 0xC70 to 0xC7C: `66 00 84 00 BF 00 FF 00 FF 00 FF 00 FF`. Since all of the values are hexadecimal, we need to convert the first register to decimal. 66 = 102. Since the values max out at FF (255 in DEC), we divide 102 by 255. 102/255 =  .4, or 40%. 
[image]
I used this math to check the entire curve and it perfectly matches. 

I then checked a register dump for when System Fan 1 was set to 100% speed in MSI Center: `FF 00 FF 00 FF 00 FF 00 FF 00 FF 00 FF`. Again, FF is full value, 100%. 

This is huge.

>Again, MSI's software pushes all these to what speed you set, but LHM doesn't.

The challenge of properly controlling the fans is an interesting one, but one I think I can program for. Since LHM doesn't have the ability to set entire fan curves, just a set PWM signal %, we can't either. Instead, the code needs to set all of the registers to the requested speed, just like MSI Center does. If people are using FanControl or some other program that hooks into LHM to control the fans, then this shouldn't be an issue since I'm _guessing_ the curves set in FanControl set the control to the requested speed on the curve every tick / refresh of the program. So, for example, if FanControl refreshes once per second, it's setting the control of the fans through LHM to a particular speed once a second. We just have to write a special case to set the entire array of registers to the desired speed, not just one register like we were thinking previously. 

As for the CPU Fan default bug, that is weird but I think we could also write a loop that just send the default command twice if needed. Janky but it'll work.

I'm going to get to work mapping all of this out on the Google Sheet, then start writing the logic to control the fans. Awesome work @demomanca !

Alcolawl, 2024-12-19:
I slapped together a test case really fast. This attempts to control System Fan 1 by setting the entire curve (All 7 Registers) to the requested speed, just as the MSI software does. This was placed in the SetControl function of Nct677Xcs:

```
                byte mode = ReadByte(FAN_CONTROL_MODE_REG[index]);
                byte bitMask = (byte)(0x01 << index);
                mode = (byte)(mode | bitMask);
                WriteByte(FAN_CONTROL_MODE_REG[index], mode);

                WriteByte(FAN_PWM_REQUEST_REG[index], 0x80); // Requests for PWM signal change
                Thread.Sleep(50);

                /* DEBUG - Case for controlling ONLY SYS_FAN 1 - SYS_FAN 1 is 3rd index, starting with 0.*/
                if (index == 2){

                    // Write fan curve
                    // Syntax is WriteByte(address,value or value in control register)
                    WriteByte(0xC70, value.Value);
                    WriteByte(0xC72, value.Value);
                    WriteByte(0xC74, value.Value);
                    WriteByte(0xC76, value.Value);
                    WriteByte(0xC78, value.Value);
                    WriteByte(0xC7A, value.Value);
                    WriteByte(0xC7C, value.Value);
                }
                else{ // Control everything else normally
                    //WriteByte(FAN_PWM_COMMAND_REG[index], value.Value);
                }
                /*END DEBUG*/
                
                //WriteByte(FAN_PWM_COMMAND_REG[index], value.Value);
                WriteByte(FAN_PWM_REQUEST_REG[index], 0x40); // Finished with request
                Thread.Sleep(50);
```
It's not pretty, nor is it meant to be, but check it out:
[image]

### It works!!!!

@demomanca You da man! But the fun doesn't stop here. We need to obviously come up with a more elegant solution, and a solution that covers all the system fans. I'm thinking we set the registers in the "FAN_PWM_COMMAND_REG" array for the system fans to the first register of the fan curve, then enter a loop that iterates 7 times, starting at the first register and incrementing by 2 so it sets all of the registers to the requested PWM value. Furthermore, we can't be sure this solution will work for other problematic X870 boards with the NCT6687D controller, so we'll have to write a conditional statement that only runs our solution for certain motherboards. The challenge there is that this work is being done within the Nct677X.cs script and that script is completely unaware of what motherboard it's interacting with. We cannot reference the Motherboard Model from within that script as the code currently stands. 

We're not out of woods yet, but this is a huge step forward to getting this done. I'm gonna hit the gym and do life stuff for a bit, then crack a cold one and sit down and try to figure some of this stuff out.

demomanca, 2024-12-19:
Good work @Alcolawl !! 

I'm playing with Argus Monitor now (which works, and funnily enough, has sys 1-6 listed backwards as well!). 
When fixing the speeds there, it also changes 0xC9F, 0xC9D, 0xC9B, 0xC99, 0xC97, 0xC95 for sys 1 to 6, but setting those still doesn't work with 0xA00 as the FAN_CONTROL_MODE_REG, so I think there must be another register fora changing the control mode for the sys fans somewhere else. I'll get argus to generate some control variations and see if I can use that to find it.

demomanca, 2024-12-19:
I'm also going to try going straight to the horses mouth. I've asked MSI via a support ticket to see if they can get someone in the bios team or the software team to tell us the addresses and commands. Can't hurt.

Alcolawl, 2024-12-19:
### I now have LHM working with FanControl. 🎉 
[image]

```
// Where FAN_PWM_COMMAND_REG = new ushort[] { 0xA28, 0xA29, 0xC70, 0xC58, 0xC40, 0xC28, 0xC10, 0xBF8 }; 

        else
        {
            // Manual mode, bit(1 : set, 0 : unset)
            // bit 0 : CPU Fan
            // bit 1 : PUMP Fan
            // bit 2 : SYS Fan 1
            // bit 3 : SYS Fan 2
            // bit 4 : SYS Fan 3
            // bit 5 : SYS Fan 4
            // bit 6 : SYS Fan 5
            // bit 7 : SYS Fan 6

            byte mode = ReadByte(FAN_CONTROL_MODE_REG[index]);
            byte bitMask = (byte)(0x01 << index);
            mode = (byte)(mode | bitMask);
            WriteByte(FAN_CONTROL_MODE_REG[index], mode);

            WriteByte(FAN_PWM_REQUEST_REG[index], 0x80); // Requests for PWM signal change
            Thread.Sleep(50);
            
            /* DEBUG - Case for controlling ONLY SYSTEM FANS*/
            if (index > 1){

                int initFanCurveReg = FAN_PWM_COMMAND_REG[index];               // Initial Register Address for the Fan Curve
                int targetFanCurveAddr = initFanCurveReg;                       // Current Fan Curve Register Address we're writing to
                ushort targetFanCurveReg = 0;                                   // Integer value of the current fan curve register address, not the value within
                int count = 0;                                                  // Countin variable because I'm dumb and can't get FOR loops to work

                // Write fan curve
                do{
                    targetFanCurveAddr = initFanCurveReg+count;
                    targetFanCurveReg = Convert.ToUInt16(targetFanCurveAddr);
                    WriteByte(targetFanCurveReg, value.Value);
                    count = count+2;
                }
                while (count < 14);
            }
            else{ // Control everything else normally
                WriteByte(FAN_PWM_COMMAND_REG[index], value.Value);
            }
            /*END DEBUG*/
            
            WriteByte(FAN_PWM_REQUEST_REG[index], 0x40); // Finished with request
            Thread.Sleep(50);
        }
```
### Explanation
The code above does all the work, most of which is in SetControl(). It's not pretty, but here's how it works:

1. User or Program sets the target fan speed through LHM, which calls SetControl().
2. SetControl() is passed two arguments: Index(Basically the fan we're targeting, 0-7) and Value (Requested %)
3. If the Index is greater than 1, it means we're targeting a System Fan.
4. initFanCurveReg is set to the register address of FAN_PWM_COMMAND_REG. (Ex. FAN_PWM_COMMAND_REG[2] = 0xC70)
5. Enter a Do-While loop where we iterate through all 7 registers for the target fan's PWM curve, setting each one to the requested speed. 
6. If the index is less than or equal to 1, behave as normal for CPU and pump fans.

Hopefully that makes sense, I didn't want to get too far into the weeds with how it works, but the point is it does. However, there are still more things to overcome.

### Good News
- We've correctly identified the Fan Speed, PWM, and Control registers
- We've also correctly identified the Fan Curve Registers
- We now have code that can read and manipulate the System Fans

### Broken Stuff
- In LHM, setting the fans to default still behaves strangely. Sometimes it won't do anything at all, and after a bit of trying different things, System Fan 1 and 2 started getting set to the same speeds. Not sure why just yet.

### Challenges
- There's a lot of bespoke code here. I have a feeling the maintainers of this project won't be a fan of that.
- We need to find a way to run this code **ONLY** for certain motherboard models, otherwise, if a motherboard has a NCT6687D, it should use the code that already exists in the 0.9.4 release
- We don't know what motherboards this will work on, or if this is an MSI-only affair. If so, this is going to get very sloppy.

Cracking a cold one shortly and the work shall continue! 🍺

Zabojnikk, 2024-12-19:
Brilliant work, gentlemen.

demomanca, 2024-12-19:
So we can, it's just a case of weather we should :)

We'll also need to figure out saving those values and returning them back when we release control. Surely there needs to be a single PWM register we can hit and have it fix that value. 

As you mentioned, given it appears that the both the model and version ID are the same, there's no way to tell what board it's on. Maybe those upper registers that are blank on the older board and not on ours (I'd like to see some more samples of this first.

Alcolawl, 2024-12-20:
Well I've come up with a way to support the MSI X870 Tomahawk without breaking support for other motherboards with the same Nuvoton controller. [I've forked the LHM repo and pushed up the commits if you want to take a look](https://github.com/Alcolawl/LibreHardwareMonitor). I also made a [release](https://github.com/Alcolawl/LibreHardwareMonitor/releases/tag/24.12.1) if people want to download and give it a shot for themselves.

I'm not entirely happy with it, but it's my best implementation so far. What I did was basically introduce a new Nuvoton Chip: NCT6687D-R, which is the actual technical model of Super I/O Controller on the X870 Tomahawk. When this motherboard, and *only this motherboard* is detected, it sets the Chip = NCT6687DR, which has it's own set of logic downstream containing the proper registers, fan control logic, etc. that we discussed above without breaking support for preexisting motherboards with the NCT6687D. 

FanControl is working wonderfully on my machine, but some of the jank and bugs still persist when attempting to control the fans from LHM directly. Setting the fans back to default only happens when closing LHM. Stuff like that is next on my list; perhaps tomorrow.

demomanca, 2024-12-20:
Awesome. I had previously done some work tidying up the voltage measurements as well, that I'd like to keep together with this. Let's use yours as the main "lets fix x870 Tomahawk" repo.

demomanca, 2024-12-20:
> Thanks for all the hard work guys. It seems MSI released a new BIOS today with the following note: "- Fix HW Monitor abnormal behavior."
> 
> Could this perhaps fix some of the issues related to the chip?

For which board? X870 Tommy is showing 7E51v1A1F released on the 13th?

Dammit if they've moved all the registers back and it works fine, gonna flip a table.....

demomanca, 2024-12-20:
Found it, X870E Carbon, sure enough, released today:

https://www.msi.com/Motherboard/MPG-X870E-CARBON-WIFI/support

Alcolawl, 2024-12-20:
> For which board? X870 Tommy is showing 7E51v1A1F released on the 13th?
>
> Dammit if they've moved all the registers back and it works fine, gonna flip a table.....

Lmao, that would be our luck. Get that table ready!

> Found it, X870E Carbon, sure enough, released today:
> 
> https://www.msi.com/Motherboard/MPG-X870E-CARBON-WIFI/support

We'll see if they drop one today for the X870 Tomahawk. I asked on OCN for any X870E Carbon owners that have flashed the new BIOS to report in on LHM readings and fan control. We'll see if I hear back. 

In other news, I've confirmed that the 0xB00 register funny business going on is in-fact, as you theorized @demomanca , the CPU Fan Curve. Right below it is the PUMP curve. What's funny about those is that the original control registers still work, I didn't have to program around those like I did the System Fans. Perhaps the 0xA28 and 0xA29 control registers get copied to the fan curve registers in a way the System Fans don't. Or maybe we haven't found the right registers that do that yet.

I'm also going to look into how setting the fans back to default works. As of right now, if you close LHM or FanControl, the fans just default back to whatever the BIOS defaults are (at least in my case, I have Smart Mode turned off for all of them). I'm not going through the trouble of reading an existing fan curve, storing it somewhere else, and then restoring it if "default" is selected, but perhaps there's a way to set the fan control back to BIOS defaults (or as MSI calls it, BIOS Mode).

poop-engineer, 2024-12-20:
Flashed the new BIOS on the X870E Carbon (forgot to mention the mainboard in the earlier post, my bad) and it doesn't seem to change anything in LHW or FanControl, sad times.

Alcolawl, 2024-12-20:
> Flashed the new BIOS on the X870E Carbon (forgot to mention the mainboard in the earlier post, my bad) and it doesn't seem to change anything in LHW or FanControl, sad times.

:( So much for that.

If you want, I can try adding support for your motherboard, if you'd like to take my fork for a spin and see if it fixes fan control. Chances are pretty high that certain things will be mislabeled, though. But this could be the first small step forward towards supporting more motherboards with the same controller and same issues.

poop-engineer, 2024-12-20:
Sure if it helps the effort, I can try to provide info and test the results. 
I'm just new to the topic so I didn't read much into how everything works yet.

Alcolawl, 2024-12-20:
> Sure if it helps the effort, I can try to provide info and test the results. I'm just new to the topic so I didn't read much into how everything works yet.

Well, in a nutshell, I believe the majority of the heavy lifting has been done. There's a lot of back and forth, but feel free to read through how the progress was made if you want to fully understand the journey and how the solution actually works. Anyways, now it's just a matter of adding the ability to identify motherboards that have the same Nuvoton NCT6687D(-R) SIO Controller and currently don't have full control over Fans and whatnot in LHM. Mind sending a screenshot of LHM on your system?

poop-engineer, 2024-12-20:
[image]
For reference: on  CPU Fan there's actually the case's (Fractal Design Meshify 2) fan splitter connected, to which the 3 fans on the AIO are connected. There's also a fan on SYS_FAN2, SYS_FAN5, and 2 fans connected via a splitter cable to either SYS_FAN4 or SYS_FAN3

Alcolawl, 2024-12-20:
Ok, lets give this a shot. I've added support for two other MSI motherboards, the PRO X870-P WIFI and your board, the MPG X870E CARBON WIFI. .Zip is attached, please follow the directions in the README. It's recommended you try LHM standalone first, check to make sure things look good (or at least look improved) before you try it with FanControl.
[LHM_More_MSIX870_v1.zip](https://github.com/user-attachments/files/18214269/LHM_More_MSIX870_v1.zip)

poop-engineer, 2024-12-21:
Thanks for your files, I paired each sensor with its speed sensor and then let them auto calibrate and it seems to work.
Only downside is that (at least in manual mode) it seems to take some time to reach the selected % and while it's getting there FanControl is reporting a question mark next to the selected %, like it's unsure where it's at currently but still somehow ends up at the right % that is selected, just a bit slower than last time I used it. 

Not sure if this has any consequences when using a curve and it taking a bit longer to ramp up/down or if I should fiddle with the %/sec ramp speeds at all (as of yet the 9800X3D has had amazing temps).

Also I'm still running the MSI bloatcenter in the background and in BIOS the fans are mostly running in smart mode, which I remember rem0o advises against, so this could also affect results at the moment. I will fiddle around with it some more and report back, but imo this is already working great. Thanks for the work.

demomanca, 2024-12-21:
Just had a look at hwmonitor outputs, which are working for temps and speeds, but not pwm speed:

```-------------------------------------------------------------------------

LPCIO Vendor			Nuvoton
LPCIO Model			NCT6687
LPCIO Vendor ID			0x5CA3
LPCIO Chip ID			0xD5
LPCIO Revision ID		0x92
Config Mode I/O address		0x4E
Config Mode LDN			0xB
Config Mode registers	
		00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 
	00	FF FF 02 FF FF FF FF 0B FF FF FF FF FF FF FF FF 
	10	FF FF 00 FF FF 0A FF FF FF FF 80 02 00 00 A2 02 
	20	D5 92 BC DF 39 FE C0 52 40 FF 8F 28 80 00 00 82 
	30	01 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
	40	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
	50	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
	60	0A 20 00 00 FF FF FF FF FF FF FF FF FF FF FF FF 
	70	00 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
Register space			LPC, base address = 0x0A20

Hardware Monitors
-------------------------------------------------------------------------

Hardware monitor		Nuvoton NCT6687
	Voltage 0		12.05 Volts [0x3EC0] (+12V)
	Voltage 1		5.00 Volts [0x3E80] (+5V)
	Voltage 2		1.32 Volts [0x5240] (CPU NB/SoC)
	Voltage 3		1.15 Volts [0x47E0] (CPU)
	Voltage 6		3.25 Volts [0xCB00] (+3.3V)
	Voltage 9		1.40 Volts [0x2BC0] (CPU VDDIO)
	Temperature 0		53 degC (127 degF) [0x35] (CPU Core)
	Temperature 1		44 degC (111 degF) [0x2C] (System)
	Temperature 2		41 degC (105 degF) [0x29] (MOS)
	Temperature 3		40 degC (104 degF) [0x28] (Chipset A)
	Fan 0			1058 RPM [0x422] (CPU)
	Fan 1			3125 RPM [0xC35] (Pump Sys 1)
	Fan 14			519 RPM [0x207] (SYS 3)
	Fan 15			962 RPM [0x3C2] (SYS 2)
	Fan 16			939 RPM [0x3AB] (SYS 1)
Hardware registers	

Hardware monitor		DIMM
	Temperature 0		41 degC (105 degF) [0x28C] (Core)

Hardware monitor		DIMM
	Temperature 0		39 degC (101 degF) [0x268] (Core)

Hardware monitor		MSI MCU
	Voltage 3		1.05 Volts [0x41B] (Chipset Core)
	Voltage 4		1.80 Volts [0x709] (CPU 1P8)
	Voltage 5		1.10 Volts [0x450] (VDD MISC)
Register space			WIN32 HID
```

mbonca, 2024-12-21:
Just installed new hardware (MSI X870 Tomahawk, AMD 9800X3D) and on frash Win11 system, and installed Fan Control and there is no speeds of SysFan1 to SysFan6. Went googling and found this website. Amazing to see there is a lot going on regarding this issue. Will keep on following. Thx guys for your hard work. Don't get this hexadecimal stuff, although I know C# code.

Alcolawl, 2024-12-21:
> Just had a look at hwmonitor outputs, which are working for temps and speeds, but not pwm speed:
>
> ```
>
> LPCIO Vendor			Nuvoton
> LPCIO Model			NCT6687
> LPCIO Vendor ID			0x5CA3
> LPCIO Chip ID			0xD5 [...]

@demomanca Those fan speed registers are completely nonsensical compared to the data and register dumps we've already gathered. Bizarre.

> Just installed new hardware (MSI X870 Tomahawk, AMD 9800X3D) and on frash Win11 system, and installed Fan Control and there is no speeds of SysFan1 to SysFan6. Went googling and found this website. [...]

@mbonca Feel free to download the [latest release](https://github.com/Alcolawl/LibreHardwareMonitor/releases/tag/24.12.3) of my LHM fork. This should get FanControl going for ya. All that I ask is you report back with good or bad news, observations, bugs of any kind, etc. The more information the better. We need to properly vet this before generating a pull request to contribute to LHM.

mbonca, 2024-12-21:
It's not working, there is no fans found, except of GPU:
[image]

Alcolawl, 2024-12-21:
> It's not working, there is no fans found, except of GPU:
> [image]
> 

The path to the sensors is incorrect. Should be "nct6687dr". Did you create a new configuration or did you try using your previous one? You have to create a new configuration and pair the speed sensors and fans manually, and you can copy over your fan curves by editing the JSON afterwards.

mbonca, 2024-12-21:
I didn't do a new configuration, used my current one. Ok, I will do as you told me to. thx 
Will keep you posted how will go.

mbonca, 2024-12-21:
How do I add fans and speed sensors manually? I can only see 2 for the GPU. There is nothing else.
[image]

Alcolawl, 2024-12-21:
Hmmm. Seems like it's not detecting the sensors. Or maybe they're hidden? 

You should be presented with something like this, just tried it on my system to confirm:

[image]

Couple with the fact that another user with the same motherboard as you had it working, I'm not sure what's going on here. Did you try opening LHM with the replaced files on your system and checking everything looks fixed?

mbonca, 2024-12-21:
App also shows app hidden fans and sensors. And I don't have LHM installed (yet).

demomanca, 2024-12-21:
Never mind, I'm a muppet, HWMonitors showing the values, not the register addresses.

ScottESanDiego, 2024-12-22:
Your nightly works good on my X870 Tomahawk, @Alcolawl .  Thanks for getting that new sensor figured out.

Smartin949, 2024-12-22:
@Alcolawl and @demomanca I can confirm that your latest release is working on the X807E Carbon with bios 7E49v1A1G. I just saw the update available to 7E49v1A21 that'll be tomorrow's project to make sure it's working correctly. Thanks for all y'alls hard work!

[87 more replies]

closed 2025-04-22 by Alcolawl