LibreHardwareMonitor/LibreHardwareMonitor #1519: NO FAN Detection on Asus z890 Strix-E

Benni231990, 2024-11-07:
Hello

I Upgrade my PC to a 285k and a z890 Strix-E but the Programm detect no Fan :(

RafaG, 2024-11-11:
Same problem here

Alcolawl, 2024-11-13:
Please attach a screenshot of LibreHardwareMonitor open so we can visualize what you're seeing. Also, it may be useful to see the same sensors in HWInfo if you have that program too. It will help identify what Super I/O controller your motherboard has and whether or not there is any underlying code to support that chip yet.

Seems as though most X870(E) motherboards are not supported or have issues, regardless of brand.

Gatoo495, 2024-11-13:
> Please attach a screenshot of LibreHardwareMonitor open so we can visualize what you're seeing. Also, it may be useful to see the same sensors in HWInfo if you have that program too. It will help [...]
[image]
Same with Aorus.

RafaG, 2024-11-24:
version 0.9.4
[image]

Gatoo495, 2024-11-25:
> @Benni231990 @RafaG Check this out #1523 Fairly sure that's going to fix your problem.

Nice one. Some of my z890 aorus pro works now. Not all. 
[image]

RafaG, 2025-01-01:
I still have the same problem, on my Asus Prime Z890M-PLUS no fan is detected. Can I contribute in any way to help solve this problem? Thanks in advance
[image]