LibreHardwareMonitor/LibreHardwareMonitor #1529: X870 AORUS ELITE WIFI7 ICE | Mainboard values and sensors are not readable.

Menayd, 2024-11-15:
Hi,

As the title says, I can't get any info from my mainboard sensors. Additionally, I'm having major issues controlling my fans via FanControl, and I suspect the issues are related.

Any help would be appreciated!

[image]

hardtimes, 2024-11-16:
Same here on Gigabyte x870 Gaming Wifi.

sebastian-dev, 2024-11-18:
Export a report and attach it here.

sebastian-dev, 2024-11-18:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1510

raubv0gel, 2024-11-26:
Nice work. But there seems to be an issue with IT87952E and fan controlling. It’s not possible to set system fan 4 to 6. Any hints where to check this in the code?

[image]

InHeritas, 2024-11-26:
> Nice work. But there seems to be an issue with IT87952E and fan controlling. It’s not possible to set system fan 4 to 6. Any hints where to check this in the code?
> 
> ![grafik](https://private-user-images.githubusercontent.com/9824146/389836557-37d0e942-9547-4dc2-9209-332281e8d78f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI2MjI1NTUsIm5iZiI6MTczMjYyMjI1NSwicGF0aCI6Ii85ODI0MTQ2LzM4OTgzNjU1Ny0zN2QwZTk0Mi05NTQ3LTRkYzItOTIwOS0zMzIyODFlOGQ3OGYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMjZUMTE1NzM1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTE4NjQ2NGZlOTg5YWUzZmJiNjYxYTQzZTJkNzAwMzUzMjFkNzNjN2FlNDk3NzVhNzUxNjYxMWY2NjRlZDNhNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.tEULZlgS48Rth1QB0i1jpKl9_OgDSI-oR_qGOrFriKU)

I'm currently using the X870e Aorus Pro Ice. The Pro and Pro Ice models only differ in color and use the same chip and layout. The % values displayed in Controls are set in the BIOS (Smart Fan). Is it not possible to operate the fans at the % values selected in Control and Manual modes from the right-click menu?

raubv0gel, 2024-11-26:
@InHeritas, exactly!

fleps, 2024-11-30:
@InHeritas @raubv0gel 
I also have the X870 Aorus PRO ICE.
Gigabyte sensors are SO BAD that not even their own FAN Control software (that instal via GCC) can see / control Fans 4 to 6!

I imagine only if they somehow fix that via BIOS would it be possible to add here?

raubv0gel, 2024-12-02:
@InHeritas, yeah LED and Fan control of GCC and UEFI is really crap. Have you ever looked at the decompiled sources of GCC? dotPeek seems to do a nice job here …