LibreHardwareMonitor/LibreHardwareMonitor #1534: Gigabyte X870 Gaming Wifi6 MB not readable.

hardtimes, 2024-11-19:
This one: https://www.gigabyte.com/Motherboard/X870-GAMING-WIFI6/support#support-manual

I see the board in libre, but no fans or temps. 

Is there anything I can provide to get this added?

Thank you.

InHeritas, 2024-11-19:
It appears that Gigabyte's new X870(E) board family uses new ITE IT8696E with ITE IT87952E. I have X870E AORUS PRO ICE and recently opened PR(https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1533) to add support for this model.

I expect the X870 GAMING WIFI6 to function similarly since it uses the same IO chip configuration. Here is the branch(https://github.com/InHeritas/LibreHardwareMonitor/tree/x870-gaming-wifi6) where I've added support for this model. If confirmed working, I'll incorporate it into my PR.

hardtimes, 2024-11-22:
I downloaded the nightly build from that branch. There are still no objects under the motherboard.