LibreHardwareMonitor/LibreHardwareMonitor #1540: Crash on KERNELBASE.dll

AbilioKID, 2024-11-24:
Just wanted to report that the latest version is crashing on **C:\Windows\system32\KERNELBASE.dll**
It seems to also complaint about **Microsoft.Win32.TaskScheduler** in .Net.
I have the latest dotnet version but my task scheduler is turned off. Not sure if it matters or not.
Went back to previous version and had no problems.

This happens on a Windows 7 Ultimate.

RokeJulianLockhart, 2025-09-27:
@AbilioKID, see https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/1151#issuecomment-1707678517.

closed 2025-09-27 by AbilioKID

RokeJulianLockhart, 2025-09-27:
> https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/1540#event-19959520375

Perhaps, reclose as unplanned.