LibreHardwareMonitor/LibreHardwareMonitor #1544: ROG CROSSHAIR X870E HERO not showing fans.

ExcessiveGBH, 2024-11-30:
Not showing any mother board fans, only GPU.

[image]

Baxtrel, 2024-11-30:
Same here, hoping it could be added so I could use fan control.
Here's a SS from HWInfo, it seems to have some ability to read the board
[image]

jtgorney, 2025-01-03:
I submitted #1581 to support the X870E Hero.

PeTeFIN, 2025-01-08:
Please add ASUS ProArt X870E-Creator Wifi as well.

1Mo-B, 2025-02-24:
I see the merged changes for the ProArt X870E-Creator Wifi but still not showing anything from the motherboard. Support would be much apprecated.

mrbrowndk, 2025-05-16:
I also have this issue on asus rog x870-i. Kindly assist