LibreHardwareMonitor/LibreHardwareMonitor #1545: Double GPU Bus

bolatulyerdos, 2024-11-30:
[image]

After deleting the name, it is restored to "GPU Bus", but changes back to "GPU Memory" after reboot.

My systems:
1. 7800X3D, B650E-E, 4060Ti
2. 2700X, B450-E, 1060

sebastian-dev, 2024-12-04:
With a 4070Ti everything looks good, try an driver update.

bolatulyerdos, 2024-12-04:
> With a 4070Ti everything looks good, try an driver update.

I have the latest version of the GPU driver.

sebastian-dev, 2024-12-04:
Attach a Report.

stray-pengo, 2025-03-20:
I'm having the same issue too.  I've messed a little with the config file, and it looks to me like a config parsing issue?

It seems that the config (name/show on gadget etc.) of the 4th entry under GPU/Load category is mirrored to that of the 5th.  The 4th is "GPU Memory" and the 5th is "GPU Bus" on all 3 of my nvidias, therefore double memory entry if the former is changed from default.

(The index in the config file is off by 1 as it counts from zero.)

The actual value is assigned and shown correctly; the issue is with the name and show/hide toggle config.