LibreHardwareMonitor/LibreHardwareMonitor #1546: Add Support VRM Temperature for ASUS ROG Strix Z590-E Gaming WiFi

Spuner, 2024-12-01:
Sorry for writing here, as I can't create "Pull requests" (the button is disabled).

I use the Fan Control application, which uses LibreHardwareMonitor, but since LibreHardwareMonitor does not show the VRM temperature, for this reason it is not shown in Fac Control either. If it is not difficult, add VRM temperature monitoring.

Thank you.

P.S. the motherboard provides access to the VRM temperature... For example, in HWiNFO the VRM temperature is displayed.

d-e-n-n-i-s-t, 2024-12-02:
Can you post a screenshot of your motherboard sensors in HWiNFO and LHM?

I don't know if I'm able to help you resolve this issue, but that would help anyone trying to help you :D

d-e-n-n-i-s-t, 2024-12-02:
The problem is that your motherboard isn't specifically supported and the default config for NCT6798D SuperIO chips for Asus is used. The problem is that the layout of voltages, temps etc isn't always accurate when a default config is used, and temps/voltages can be missing

I don't really have the time right now to write the code for you. However, if you really want proper support for your board, you could follow the approach I took in #1532 for my own board, or wait until someone else is willing to help you

d-e-n-n-i-s-t, 2024-12-02:
The good thing is that there a some other motherboards from Asus that use the same IO chip, and they seem to have very similar layouts. Copying those for your motherboard would probably work

Spuner, 2024-12-02:
For example, which ones?

P.S. To be honest, I looked at [#1532](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1532), and I don't quite understand what and where to add...