LibreHardwareMonitor/LibreHardwareMonitor #1551: No CPU temperature data on Lenovo P7 with Xeon W5-3435X on Windows Server Core 2019

alexkmbk, 2024-12-03:
Lenovo P7
OS:  Windows Server Core 2019
CPU: Xeon W5-3435X

MaHoBo, 2024-12-04:
please create a report and upload or paste it here.

alexkmbk, 2024-12-04:
> please create a report and upload or paste it here.

When I try to save the Report  I have got an error:
************** Exception Text **************
System.Runtime.InteropServices.COMException (0x80040111): Creating an instance of the COM component with CLSID {C0B4E2F3-BA21-4773-8DBA-335EC946EB8B} from the IClassFactory failed due to the following error: 80040111 ClassFactory cannot supply requested class (Exception from HRESULT: 0x80040111 (CLASS_E_CLASSNOTAVAILABLE)).
   at System.Windows.Forms.SaveFileDialog.CreateVistaDialog()
   at System.Windows.Forms.FileDialog.RunDialogVista(IntPtr hWndOwner)
   at System.Windows.Forms.CommonDialog.ShowDialog(IWin32Window owner)
   at LibreHardwareMonitor.UI.MainForm.SaveReportMenuItem_Click(Object sender, EventArgs e)
   at System.Windows.Forms.ToolStripItem.RaiseEvent(Object key, EventArgs e)
   at System.Windows.Forms.ToolStripMenuItem.OnClick(EventArgs e)
   at System.Windows.Forms.ToolStripItem.HandleClick(EventArgs e)
   at System.Windows.Forms.ToolStripItem.HandleMouseUp(MouseEventArgs e)
   at System.Windows.Forms.ToolStrip.OnMouseUp(MouseEventArgs mea)
   at System.Windows.Forms.ToolStripDropDown.OnMouseUp(MouseEventArgs mea)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ToolStrip.WndProc(Message& m)
   at System.Windows.Forms.ToolStripDropDown.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

[image]

MaHoBo, 2024-12-08:
please try pull request #1555 and provide some feedback about temperature support.
hard to test without having such CPU at hand ;)

alexkmbk, 2024-12-09:
> please try pull request #1555 and provide some feedback about temperature support. hard to test without having such CPU at hand ;)

Yes, I have just tested it and it works now, thank you!