LibreHardwareMonitor/LibreHardwareMonitor #1574: Read Rate and Write Rate

Jack-InGitHub, 2024-12-26:
When I try to get Value by sensor.Value method from LibreHardwareMonitorLib.dll , everything is output well, except for the speed of writing and reading disks.

A strange value is output that does not change when the disk is loaded. In the exe program, the information is displayed correctly.

**LibreHardwareMonitor.exe the output is as follows.**

[image]

```python
import os
import clr #pythonnet

if __name__ == "__main__":
    clr.AddReference(os.getcwd() + "\\LibreHardwareMonitorLib.dll")

    from LibreHardwareMonitor import Hardware

    handle = Hardware.Computer()
    handle.set_IsStorageEnabled(True)
    handle.Open()
    
    for hardware in handle.Hardware:
        print(f"hardware---------->{str(hardware.HardwareType)}")
        hardware.Update()
        for sensor in hardware.Sensors:
            print(f"\tsensor---------->{str(sensor.SensorType), sensor.Name, sensor.Value}")
```

**When running with administrator privileges, the output is as follows.**

```shell
hardware---------->Storage
        sensor---------->('Temperature', 'Temperature', 48.0)
        sensor---------->('Level', 'Available Spare', 100.0)
        sensor---------->('Level', 'Available Spare Threshold', 10.0)
        sensor---------->('Level', 'Percentage Used', 1.0)
        sensor---------->('Data', 'Data Read', 44793.0)
        sensor---------->('Data', 'Data Written', 18058.0)
        sensor---------->('Temperature', 'Temperature 1', 47.0)
        sensor---------->('Temperature', 'Temperature 2', 42.0)
        sensor---------->('Load', 'Used Space', 74.89056396484375)
        sensor---------->('Load', 'Read Activity', 4.116503259865567e-05)
        sensor---------->('Load', 'Write Activity', 9.133057574217673e-06)
        sensor---------->('Load', 'Total Activity', 99.99571990966797)
        sensor---------->('Throughput', 'Read Rate', None)
        sensor---------->('Throughput', 'Write Rate', None)
```

'Load', 'Total Activity',
'Throughput', 'Read Rate',
'Throughput', 'Write Rate',
**The data output from these three items are all incorrect.**

Jack-InGitHub, 2024-12-26:
I read issue 683, but my problem doesn't seem to be solved.
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/683

mkreuzmayr, 2025-05-20:
Hey @Jack-InGitHub, I'm also encountering a similar issue when trying to access disk activity/throughput from the DLL in a different application context. All other sensor data seems to be working, but those specific values for storage activity are missing for me as well. It's interesting as the official LHM application for me also works correctly.

Have you had any luck investigating this further since your initial post?

jlj001031, 2025-06-06:
i encountered the same problem when i use the dll, how can i resolve it

Jack-InGitHub, 2025-06-06:
> Hey [@Jack-InGitHub](https://github.com/Jack-InGitHub), I'm also encountering a similar issue when trying to access disk activity/throughput from the DLL in a different application context. All [...]

With no better options, I dropped the parameters.