LibreHardwareMonitor/LibreHardwareMonitor #1576: C API

thewh1teagle, 2024-12-27:
Hello,
Thank you for your work on this library! I would like to use it from a Rust project, and having a C API would make integration much easier. Would it be possible to expose a C API for this library?

Thanks

baoyifeng, 2024-12-30:
you can Try using C++/CLI.

baoyifeng, 2024-12-30:
you can try using C++/CLI.
I use it from a c++ project.

#include <iostream>
#include <string>
#include <windows.h>

#include <msclr\marshal_cppstd.h>
using namespace std;

// LibreHardwareMonitorLib.dll
// https://productionresultssa19.blob.core.windows.net/actions-results/b0d9359e-28e4-4a09-bb39-2b59492cb501/workflow-job-run-ca395085-040a-526b-2ce8-bdc85f692774/artifacts/5ac5bb08f98808ee429c3d9f858129d3c0e3a65869d320d5670ea7985656abdf.zip?rscd=attachment%3B+filename%3D%22LibreHardwareMonitor-net472.zip%22&se=2024-12-23T06%3A22%3A01Z&sig=nEQ1pcCHcbfq%2BHMD5rAGB0avNElMX0rTGTDw50WTub8%3D&ske=2024-12-23T15%3A02%3A00Z&skoid=ca7593d4-ee42-46cd-af88-8b886a2f84eb&sks=b&skt=2024-12-23T03%3A02%3A00Z&sktid=398a6654-997b-47e9-b12b-9515b896b4de&skv=2024-11-04&sp=r&spr=https&sr=b&st=2024-12-23T06%3A11%3A56Z&sv=2024-11-04
#using "E:\2\LibreHardwareMonitor-master\bin\Debug\net481\LibreHardwareMonitorLib.dll"
using namespace LibreHardwareMonitor::Hardware;
using namespace LibreHardwareMonitor;
int main()
{
    
    Computer^ computer = gcnew Computer();

    computer->Open();
    //computer->IsBatteryEnabled = true;
    computer->IsControllerEnabled = true;
    computer->IsCpuEnabled = true;
    computer->IsGpuEnabled = true;
    computer->IsMemoryEnabled = true;
    computer->IsMotherboardEnabled = true;
    //computer->IsNetworkEnabled = true;
    //computer->IsPsuEnabled = true;
    //computer->IsStorageEnabled = true;
    //auto aaa = gcnew UpdateVisitor();
    //computer->Accept(gcnew UpdateVisitor());

    for each(auto hardware in computer->Hardware)
    {
        std::string name = msclr::interop::marshal_as<std::string>(hardware->Name);

        std::cout << "name:" << name << std::endl;
        //Console.WriteLine("Hardware: {0}", hardware->Name);

        for each (auto subhardware in hardware->SubHardware)
        {

            std::string SubHardware = msclr::interop::marshal_as<std::string>(subhardware->Name);
            std::cout << "SubHardware:" << SubHardware << std::endl;

            //Console.WriteLine("\tSubhardware: {0}", subhardware.Name);

            for each(auto sensor in subhardware->Sensors)
            {

                std::string name = msclr::interop::marshal_as<std::string>(sensor->Name);
                std::cout << "Sensors:" << name << std::endl;
                //Console.WriteLine("\t\tSensor: {0}, value: {1}", sensor.Name, sensor.Value);
            }
        }

        std::cout << "==============================" << std::endl;

        for each(auto sensor in hardware->Sensors)
        {
            std::string name = msclr::interop::marshal_as<std::string>(sensor->Name);
            std::string SensorType = msclr::interop::marshal_as<std::string>(sensor->SensorType.ToString());
            std::string Identifier = msclr::interop::marshal_as<std::string>(sensor->Identifier->ToString());
            std::string Parent = msclr::interop::marshal_as<std::string>(sensor->Hardware->Identifier->ToString());

            float v = 0;
            if (sensor->Value.HasValue)
            {
                v = sensor->Value.Value;
            }
            
            //std::string v = msclr::interop::marshal_as<std::string>(sensor->Value);
            std::cout << "name=" << name << ",SensorType=" << SensorType << ",Identifier=" <<Identifier << ",Parent=" << Parent ;
            std::cout << "value=" << v <<std::endl;
            //Console.WriteLine("\tSensor: {0}, value: {1}", sensor.Name, sensor.Value);
        }
    }

    computer->Close();
    return 0;
}

baoyifeng, 2024-12-30:
c++/cli is hard to Maintain and expand the code. I'm going to rewrite a c# program to write the data to wmi.Other programs will pull data from the wmi.

chmichael, 2024-12-30:
@baoyifeng wmi is deprecated and slow

baoyifeng, 2025-01-02:
> @baoyifeng wmi is deprecated and slow

I don't access data very often. if slow，I’ll try other ways, like sqlite or shared memory.

clach04, 2025-01-12:
> c++/cli is hard to Maintain and expand the code. I'm going to rewrite a c# program to write the data to wmi.Other programs will pull data from the wmi.

Why not use the existing WMI support?

  * https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/tree/master/LibreHardwareMonitor/WMI
  * https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/1032#issuecomment-1497724246 shows this working

clach04, 2025-01-12:
> Hello, Thank you for your work on this library! I would like to use it from a Rust project, and having a C API would make integration much easier. Would it be possible to expose a C API for this library?
> 
> Thanks

Existing rust project https://github.com/Levminer/cores