LibreHardwareMonitor/LibreHardwareMonitor #1577: CPU Temperature Sensor

tessierp, 2024-12-29:
Trying to get the my 7800X3D temperature sensor. Only 1 is returned and the value is always 0 which I find very odd. Could that be a bug? I tried the application that you guys made and temps are shown so not sure if this is a bug or perhaps you calculate it manually?

jansteffen, 2024-12-30:
5800X3D here, it's not showing me CPU temps either:

[image]

sebastian-dev, 2024-12-31:
This is correct, the cpu has one temperature sensor. Additional sensors are only available via SMU.

tessierp, 2025-01-02:
> This is correct, the cpu has one temperature sensor. Additional sensors are only available via SMU.

I'm not sure I understand.. I'm looking for the sensor type temperature, what is returned to me is a value of 0 which isn't right.

sebastian-dev, 2025-01-02:
Post a Report (File).
It looks like the driver is not installed correct. Maybe blocked from a AV soulution.

sebastian-dev, 2025-01-03:
Application (Menu) File->Save Report.
Attach this file here.

tessierp, 2025-01-04:
> Application (Menu) File->Save Report. Attach this file here.

But wait I'm not using the application, I'm using the API to output data to ZABBIX and as I look through all the available sensors and the data that comes back has no temperature sensor.

tessierp, 2025-01-15:
Hello? Any clue anyone?

sebastian-dev, 2025-01-16:
Use the application (LHM) and check whether the data is available there.
You can also create a report from the application.
If there are no problems, the problem lies with the other software.

tessierp, 2025-01-17:
> Use the application (LHM) and check whether the data is available there. You can also create a report from the application. If there are no problems, the problem lies with the other software.

As I explained, I'm using HWINFO, AMD Adrenaline as well as many others they all report the temperatures.. I also tried the application made by your team and yes the temperature is displayed.  But that is NOT THE ISSUE. The issue is when I use the LibreHardwareMonitor's libraries to crawl through all the sensors available, the temperature sensor is not in the list!! Someone else also reported that LHM is not reporting temps.

So I guess I'll have to show you what I am doing.

First I get the details of my CPU 

```
        static IEnumerable<CpuData> GetSystemCPUInfo()
        {
            Computer computer = OpenComputer();

            try
            {
                computer.Accept(new UpdateVisitor());

                var cpuData =
                    (from hw in computer.Hardware
                     where hw.HardwareType == HardwareType.Cpu
                     select new CpuData(hw)).ToList(); ;

                return cpuData;
            }
            catch (Exception ex)
            {
                Console.WriteLine(ex.ToString());
                throw;
            }
        }
```

Then, from the data that is returned I enumerate all the sensors

```
                var cpuDataTest = GetSystemCPUInfo();
                foreach (ISensor sensor in cpuDataTest.FirstOrDefault().Sensors)
                {
                    Console.WriteLine($"\tSensor: {sensor.Name}, type {sensor.SensorType.ToString()}, value: {sensor.Value}");
                }
```

and the list of sensors I already gave you from what I wrote above. So my question again is WHY is the temperature sensor not shown?

Data-Corruption, 2025-01-18:
Hello, I had the same issue, running my exe as admin fixed it, hope it helps.

tessierp, 2025-01-18:
> Hello, I had the same issue, running my exe as admin fixed it, hope it helps.

Well that fixed my issue, running Jetbrain Rider in admin and debugging the application returned a value for the temperature sensor! 

Thanks @Data-Corruption !

closed 2025-01-18 by tessierp