LibreHardwareMonitor/LibreHardwareMonitor #1582: MSI MPG Z890 EDGE TI WIFI not showing fans

Uwe-12, 2025-01-04:
Not showing any mother board fans, only GPU and CPU. Hoping it could be added so I could use fan control.
In HWInfo the fans are shown correct.
[image]

demomanca, 2025-01-05:
Yeah, we're tackling this in issue #1513 (we hope)

Alcolawl, 2025-01-24:
Hey @Uwe-12 , if you'd like I can try to get LHM working with your motherboard but it'll require some of your cooperation. 

Open an issue on [my fork of LHM ](https://github.com/Alcolawl/LibreHardwareMonitor)and we'll work it out. The hope is to get this to a place where we can create a pull request and get it merged with the Master LHM repo for public release.

Uwe-12, 2025-01-24:
Hallo Tyler

Thank you for your effort. I've open an issue on my fork of LHM
<https://github.com/Alcolawl/LibreHardwareMonitor> 

I hope that this can help you to find a solution.

 

Uwe-12

TI WIFI not showing fans (Issue #1582)

 

<https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/1582#is
GKUNAVCNFSM6AAAAABUTAD5PSVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDMMJRGM3DC
NRYGQ> .
<https://github.com/notifications/beacon/BOFYVQ2O6PDEFDS5T3KSGFD2MGKUNA5CNFS
M6AAAAABUTAD5PSWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTU
hub.com> >

Uwe-12, 2025-01-24:
[HWMonitor.txt](https://github.com/user-attachments/files/18536335/HWMonitor.txt)

[LibreHardwareMonitor Report.txt](https://github.com/user-attachments/files/18536341/LibreHardwareMonitor.Report.txt)