LibreHardwareMonitor/LibreHardwareMonitor #1589: some settings are not persisted/ dont survive a restart of the app

elonderin, 2025-01-09:
some settings are not persisted/ dont survive a restart of the app.

I have noticed these but there might be more:
- plot > value axis > auto scale
- menu > options > file rotation method

It seems that they are always reset to the 1st entry

BTW: i have unzipped it on `D:\bin\_hardware\LibreHardwareMonitor-net472` so i'm not suffering from #995

orthello, 2025-04-18:
added a pull request to make the FileRotationMethod setting persistent

jez9999, 2025-09-04:
Actually it looks like it has been fixed in this commit:

https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/commit/cb3993814305f81b5e2c3b827bb39e85065dfb9b

But it hasn't been released as a final version yet...