LibreHardwareMonitor/LibreHardwareMonitor #1594: No fans detected on Asus Prime X870-P

sabifa, 2025-01-11:
Hey all,
after installing LHM / FanControl on my new system I unfortunately noticed that no fans are detected at all. No CPU fan, no chassis fan - nothing. In fact, there are no sensor readings of the entire motherboard.

I checked with HWMonitor to see if there is an issue with my hardware but there I can see all fans:
[image]

I am not sure if this is related to other X870 (E) posts like #1513 #1490 #1510 or #1488 

If there is any way to help please let me know.

SunnyStefan, 2025-01-13:
This issue is impacting several AMD 8XX series motherboards. In addition the ones you linked at the end of your post, it's also a problem with the ASUS B850-I motherboard.

sabifa, 2025-01-13:
Is there any way to help add support for these boards?
I can't find an explanation in the readme on what is needed to support new motherboards.

bbnn12, 2025-01-13:
just wann add to this also currently a problem with MSI x870 Tomahawk, CPU header works tho, just not system fan headers, cant confirm pump header

rrodriguez81, 2025-01-19:
Same problem here with the same motherboard, a MSI MAG X870 Tomahawk WIFI. Thanks for your work.

Komalis, 2025-01-27:
JHaving the same with a MSI PRO B850-P Wifi

sabifa, 2025-03-08:
I've reinstalled FanControl and it works with the latest version, thx

closed 2025-03-08 by sabifa

Zaubergnom, 2025-09-10:
I have downloaded the latest overnight version and I still do not get any values of the fans. 
Board: ASUS Prime X870-P, 
Controller: Nuvoton6701D.
Help would be appriciated