LibreHardwareMonitor/LibreHardwareMonitor #1597: No sensors detected on ASUS TUF Gaming B850M-PLUS

dpri, 2025-01-14:
As the new B850 chipset from AMD is IIRC more or less a rebranded B650E, perhaps the inclusion of that new chipset shouldn't be as big a problem. If I can give further information or help with testing, please let me know.
[image]

Wyllio, 2025-01-14:
This has been an issue since the release of X870 and Z890 in September 2024. Still most of the new motherboards use the Nuvoton controller.

dpri, 2025-01-19:
As of today, it works. 
Thanks a lot, LHM team!

[image]

closed 2025-01-19 by dpri