LibreHardwareMonitor/LibreHardwareMonitor #1610: No data is detected under the sensor Nuvoton NCT6701D(ROG STRIX B850-F GAMING WIFI)

ezraylcgithub, 2025-01-21:
After upgrading to the latest version, the sensor model can be detected, but no data can be detected.(latest nightly version)
Is there something I should set up to get this right? 
Or if I can give further information or help with testing, please let me know.

[image]

morrie-morrie, 2025-01-28:
I'm seeing the same thing. Let me know if there is anything I can do to assist.

ezraylcgithub, 2025-02-08:
Finally found the reason，just turn off the Fan Xpert control switch in the Armoury Crate app on ASUS motherboards and restart windows.

closed 2025-02-08 by ezraylcgithub

morrie-morrie, 2025-02-08:
> Finally found the reason, just turn off the Fan Xpert control switch in the Armoury Crate app on ASUS motherboards and restart windows.

I tried this and did not get the same result. I still see the same missing data left at Nuvoton NCT6701D.

Did you uninstall Armoury Crate.
I'm running the current nightly build.

Andrew.

ezraylcgithub, 2025-02-09:
> I tried this and did not get the same result. I still see the same missing data left at Nuvoton NCT6701D.
>
> Did you uninstall Armoury Crate.
> I'm running the current nightly build.
>
> Andrew.

Maybe you can disable it to boot up

Artorus3333, 2025-02-26:
For me, it helped to deactivate and delete the process AsusFanControlService.exe