LibreHardwareMonitor/LibreHardwareMonitor #1622: Crucial Ballistix temperature

TheDatadrainer, 2025-01-29:
Hi.
Would it be possible to add temperature monitoring for Crucial Ballistix DDR3 & DDR4 memory modules?
A tool from Crucial named Ballistix MOD can do this, but it is abandoned and obsolete, and it would be great to have a support in LHM as there is no support anywhere else.
Can you please consider it? Thank you.

Blacktempel, 2026-01-01:
There is no more `Generic Memory` in LHM, unless you are using 0.9.4 (the latest release) - which does not use RAMSPDToolkit at all as it was introduced later.
Please check with a recent master build from ["Actions" tab.](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions)

If it works with `ConsoleOutputTest`, it also works with LHM.

Happy new year :)

Blacktempel, 2026-01-01:
The RAM modules should be below Virtual and Total Memory. One for each stick.

Blacktempel, 2026-01-01:
The WinRing0 version in RAMSPDToolkit is self-contained, does not install anything & you can remove the whole folder afterwards. Would have been interesting to also see what this can do. That SMBus implementation contains a bit more logging as the other one is written in Pawn and just being called from C# code.
(May still require a temporary exception from Windows Defender, unfortunately)

Shouldn’t this be the i801 SMBus for the RAM modules ? Or is there something special for your hardware configuration ?

If it doesn’t work with WinRing0 or PawnIO, the issue is something else. Or maybe both just lack the required implementation.

If it works with WinRing0 and not PawnIO, the log from the WinRing0 version could be helpful.

TheDatadrainer, 2026-01-02:
Here is the log from the WR0 version:
- [Log.txt](https://github.com/user-attachments/files/24407857/Log.txt)

I don't know if the chipset of my motherboard is compatible but theoretically it should be. It as a chipset Intel X299 codename Basin Falls, and in the Linux kernel, it is supported by the i801 driver, DEVID A2A3 [https://github.com/torvalds/linux](https://github.com/torvalds/linux/blob/master/drivers/i2c/busses/i2c-i801.c). And the logs from both version of ConsoleOutputTest.exe shows it is correctly detected and used addresses are listed. Apart from that, I don't know... I'm quite lost here but I'll try to help as much as I can.

DavidTobar456, 2026-01-02:
Hey, here from the RAMSPDToolkit team (basically just BlackTempel and myself), what you're saying is odd to say the list, in terms of having the addresses.

First things first, it's a first time for me seeing that there are two RAM modules with the very same SMBus Address, I checked the output you got us from CPU-Z to see if we can gain any valuable info from it, such as there being two modules, but only one responding to SMBus transactions (this would mean DIMM 2 and DIMM1 should share the same values in the SPD registers table).

But apparently they're not the same, so, from an electronics perspective, and understanding how the I2C/SMBus protocol works what would make sense is for there to be two parallel connections into the SMBus muxed (this would mean just one is connected at a time). If this were to be the case, the system should do "something" to tell the MOBO which DIMM slot will be connected into the SMBus if any were to be connected. Now, if my intuition is right, and one were to not set any of the two lines to be read, none would appear in the SMBus and hence none would be detected. So unless we knew what that "something" is, we'd be left with nothing to work with.

Do take into consideration this is only my intuition, and I have to dig deeper into the subject to see if there's something that's not related at all to the SMBus Controller. I'll be looking into that. But, as far as detection logic goes, it is standardized by JEDEC, so we shouldn't have any issues with it in RAMSPDToolkit.