LibreHardwareMonitor/LibreHardwareMonitor #1626: Ryzen 9 9900X clock frequencies are wrong

John1956NL, 2025-02-04:
Hi,

I really like your dll and it has been helpful with many CPU I track, most all are Intel. However, I find that on my personal Ryzen 9 9900X, the clock frequencies are wrong. The image below shows the various cores and their frequencies.

[image]

KeithMyers, 2025-02-09:
Yep, the MSR register numbers obviously changed a lot for Family 1Ah cpus like the 9900X or 9950X.
This is what zenmonitor3 root access shows for my 9950X after I modified the Zen3 family ID define to 1Ah so that the program finally recognized the 9950X as a Zen cpu.  Before doing that the program simply stated "No ZEN cpu found" when launched.

My core clocks are wonky too when the program is launched in root mode to read the power MSR registers.  But when launched in standard userland mode and not root, the core clocks are correct.

Root mode.

[image]

Standard userland mode.

[image]

einzel, 2025-03-24:
I'm having a similar but different problem. All of my cores are shown as full speed even when the CPU is idling as shown in HWinfo and even taskman.