LibreHardwareMonitor/LibreHardwareMonitor #1634: Support for MSI B850 Gaming Plus WiFi is incomplete for fans detection

mjouve-jul, 2025-02-10:
Hello,

I created a first issue in a Fork of this repo and just realized my mistake.

Subject says it all, like a lot of recent MBs, system fans are not detected for the MSI B850 Gaming Plus WiFi, only the PUMP, CPU and GPU are.

I have copy/pasted the rest of my initial issue below with all details available.

Thank you 

> First time ever posting an issue on GitHub so my apologies if everything isn't posted as expected. 
> 
> I think I have searched everywhere in this repo and couldn't find anything related to the support of MSI B850 Gaming Plus WiFi. 
> 
> Like a lot of users recently, SYS_FANS are not showing neither in Fan Control or LHM, though they are detected by HW Monitor. See screenshot below for proof : 
> 
> [image]
> 
> I made sure to follow instructions first and copy the 3 files (dll, pdb, xml) to the LHM & Fan control folders and then restart Fan Control and create a new profile but it didn't work, unfortunately.
> 
> Also attached, LHM report as well as the HW Monitor one :
> 
> [LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/18718973/LibreHardwareMonitor.Report.txt)
> 
> [HWMonitor.txt](https://github.com/user-attachments/files/18718981/HWMonitor.txt)
> 
> I see you are very pro-active on these issues and I thank you very much for that, your hard work is greatly appreciated.
> 
> Cheers

Slickrick17, 2025-02-21:
There's a pending PR for this support: [Add Support for MSI AM5/LGA1851 Motherboards](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1644).  Have you tried copying the 3 LHM files from that branch?  

In case you're unable to compile it, here they are:

[LHM - MSI AM5 LGA1851 Motherboards.zip](https://github.com/user-attachments/files/18915830/LHM.-.MSI.AM5.LGA1851.Motherboards.zip)

mjouve-jul, 2025-02-22:
Indeed, Alcolawl provided the files in another issue I created under his fork and the fans are now detected !