LibreHardwareMonitor/LibreHardwareMonitor #1642: CPU temperature isn't available with Sample code

sms2000, 2025-02-13:
When I build and run the sample code I can see GPU temperature. But CPU temperature values are always 'null'.
I can see all Core Temperature sensors but all TjMax sensors but the values are 'null'.

The LibreHardwareMonitor application shows all CPU Temperatures correctly.

Any ideas what should be configured in 'Computer'?

frogcrush, 2025-02-25:
Hello, this sounds like you are not running your application as administrator as per the README.

If this is resolved, please close this issue.

closed 2025-02-26 by sms2000

MrRStarkey, 2025-11-02:
> When I build and run the sample code I can see GPU temperature. But CPU temperature values are always 'null'.
> I can see all Core Temperature sensors but all TjMax sensors but the values are 'null'.
> 
> The LibreHardwareMonitor application shows all CPU Temperatures correctly.
> 
> Any ideas what should be configured in 'Computer'?
> 
> EDIT: resolved. My fault.

Can you inform me of the solution? I am experiencing the same problem and have ran the compiled file as admin but still not getting all of the sensor data that I see running the full app.

sms2000, 2025-11-02:
> Can you inform me of the solution? I am experiencing the same problem and have ran the compiled file as admin but still not getting all of the sensor data that I see running the full app.

Sorry, it was long ago. We found this lib useless for our project (lacks too many features).

AFAIR I compiled the lib incorrectly, disabling Ring0.