LibreHardwareMonitor/LibreHardwareMonitor #1648: Add metrics endpoint for prometheus

Lifailon, 2025-02-21:
Tell me, is it possible to add an endpoint for receiving `metrics` in prometheus format?

At the moment, when starting a web server, it is possible to receive data in `json` format, but this requires an additional exporter to parse the output, is it possible to integrate this into the api or add an additional button (a separate api server)?

Now I am thinking about the implementation, so as not to start the web server, you can parse logs and collect metrics directly in the container via wsl, here is an example of access:

`/mnt/c/Users/Lifailon/Documents/LibreHardwareMonitor-0.9.4/LibreHardwareMonitorLog-2025-02-21.csv`

> I planned to use this as a temporary solution...

or create an application that will process data via `WMI` in the system itself.

I can offer to implement the server as a separate executable file that could be included in the release and started / stopped through an interface similar to the web server. Will you be able to accept this in the future and integrate it into the app?

leandrofus, 2025-05-13:
I had added this functionality here, but the MR was rejected.
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1627

kryztoval, 2025-09-05:
The reason quoted in the rejection was "This adds another dependency"

It should be trivial to make it so the _root and all of its children are transversed and formatted accordingly to OpenMetrics without adding any extra dependencies.

I can do that, what I have not been able to figure out is what is the logical way to make the sensors in prometheus.
For example, in LHM "Total Memory" and "Virtual Memory" have the same "Load > Memory" metrics.

Should they override each other? are they part of a different dimension? are they in the same group? should they be virtual_memory_load_memory or virtual_memory_memory_load and total_memory_memory_load ? does this rule work with all sensors?

Help me out with that and I can make a request that adds no dependencies that exposes a /metrics endpoint for prometheus.

kryztoval, 2025-10-19:
Soon you should be able to see `http://ip/metrics` directly from the web server.

What grafana dashboard are you using?

closed 2025-12-24 by PhyxionNL