LibreHardwareMonitor/LibreHardwareMonitor #1657: Running LibreHardwareMonitor blocks ejection of removable media

Korkman, 2025-03-08:
As the title says. Without digging into the code, I would assume some kind of handle for the storage devices is kept open between polls and that should not be the case to allow ejection.

azlux, 2026-10-24:
Are you sure about that ?
I've run the Process Monitor on windows, and found out too than LibreHardwareMonitor always try to create file on my removable media.
That block me to eject it.

[image]
version 0.9.7-pre717

Blacktempel, 2026-10-26:
@azlux 
You can disable USB storage device monitoring in settings since #2560
It is necessary to open the device to directly query the necessary data.

If you still think this is a bug, please open an issue in [DiskInfoToolkit](https://github.com/Blacktempel/DiskInfoToolkit).
If you are able to reproduce the behaviour and work with C#, help is always appreciated.

---

Edit: please always test with a recent version.

> version 0.9.7-pre717

we are currently on 0.9.7-pre772

azlux, 2026-10-26:
Hi @Blacktempel,
Thanks for the fast answer.
It's indeed all good with the new version.
The option to avoid USB device is working well