LibreHardwareMonitor/LibreHardwareMonitor #1660: HackTool:Win32/Winring0 is blocked by Windows Defender today

alski, 2025-03-11:
Although this feels like a classic antivirus false positive, can we get confirmation that all commits in the 0.9.4 release did not introduce any malicious changes?

For anyone else coming across this 

[image]

I'm developing with LibreHardwareMonitorLib 0.9.4 and came down to a hanging VisualStudio, but a running application, and the above fun message.

alski, 2025-03-11:
FYI many users reporting on on https://github.com/Rem0o/FanControl.Releases/discussions/3017 for multiple products

5FeetUnder, 2025-03-11:
there's also [this comment](https://github.com/Rem0o/FanControl.Releases/issues/3016#issuecomment-2713558302) by @malikm on Rem0o/FanControl.Releases#3016:
> This is due to a vulnerability (or rather multiple ones) in the WinRing0 driver that is known for many years. All vendors were aware of this long ago but didn't perform the required (rather extensive) changes. Besides the need to significantly rewrite the kernel driver, application and interface between them, it also requires a new digital signature that's quite expensive for FOSS projects and can be issued only to a business (the signing needs to be done via MS HW/WHQL site).
> Microsoft was aware of this vulnerability and started tightening rules long ago. It also notified respective vendors about an upcoming full blocking of this driver. Initially it was planned to happen in 2024, then Jan'25, and now it seems they finally did it.
> There's no other way around other than rewriting the driver from scratch to be reliable, robust and secure. A lot of effort...

electricJP, 2025-03-11:
So, what does that mean? Project is dead and we need to move to something else?

chilman408, 2025-03-11:
Same here.  What is the recommended action for people who is using this?

clementbey, 2025-03-11:
Same here, 10 of our servers are affected.

gringrant, 2025-03-12:
I took the time to make a write up & explanation w/ sources of the WinRing0 vulnerability (in the context of impacting Fan Control) on Reddit:

https://www.reddit.com/r/FanControl/comments/1j93doq/why_does_defender_hate_fan_control_an_explanation/

You can also just read my sources directly:

https://medium.com/@matterpreter/cve-2020-14979-local-privilege-escalation-in-evga-precisionx1-cf63c6b95896
https://nvd.nist.gov/vuln/detail/cve-2020-14979
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/984

globi84, 2025-03-12:
I have the same problem; I use this library to monitor the temperature of our workstations and servers.

tsoriano64, 2025-03-12:
Hi, I've been using this library for years on a personal project, and I've been getting this alert all the time since yesterday.
Is this going to be fixed, or is there an alternative to this library?

MrAnnonymous00, 2025-03-12:
After Updating Defender(KB2267602, ver1.423.350.0), it's not blocked anymore. But it could be temporary pullback. There's still possibility that MS may block WinRing0 again sooner or later.

> https://github.com/Rem0o/FanControl.Releases/issues/3016#issuecomment-2718171928 :

> Classic Microsoft pullback, though it's only a matter of time before it gets blocked again.

PhyxionNL, 2025-03-17:
This is indeed unfortunate, but this is more or less a dupe of [#1656 ](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/984). Closing so all can comments can be in one topic.

closed 2025-03-17 by PhyxionNL