LibreHardwareMonitor/LibreHardwareMonitor #1661: Gigabyte B550M Aorus Elite

lightsparkk, 2025-03-11:
Libre Hardware Monitor is unable to read information and also makes Fan Control unusable.

[image]

[image]

tiagolo, 2025-03-11:
Same issue on an ASROCK B550 Steel Legend.

[image]

tiagolo, 2025-03-11:
After allowing the file `LibreHardwareMonitor.sys` on Windows Defender it worked again.

My windows event log had the following entry:

```
Erro	11/03/2025 20:02:49	Service Control Manager	7000	Nenhum
Não foi possível iniciar o serviço R0LibreHardwareMonitor devido ao seguinte erro: 
Não foi possível concluir a operação com êxito porque o arquivo contém um vírus ou software possivelmente indesejado.
```

Sorry for the portuguese logs. :D