LibreHardwareMonitor/LibreHardwareMonitor #1686: Nvidia Memory Junction Temperature Wrong on RTX 5090

F3NN3X, 2025-04-01:
Recently noticed that LHM is reporting the wrong temp of the memory junction on my 5090,
i don't know if this is a problem with other 5xxx cards.

LHM reports the temp as 255*

Card is MSI GeForce RTX 5090 VENTUS 3X OC

F3NN3X, 2025-04-07:
this is probably why: https://videocardz.com/pixel/nvidia-has-removed-hot-spot-sensor-data-from-geforce-rtx-50-gpus

efeurun, 2025-04-15:
I'm having the same issue. I thought something was wrong with my sensors.

majort94, 2025-05-03:
I can confirm this issue with my Zotac 5090 Amplify. Notably, GPU-Z seems to have the correct "Memory temperature", but does not have a hotspot temperature, as referenced above.

PangciPG, 2025-05-04:
I can confirm this issue with my 5070Ti Maxsun.

Betalos, 2025-09-19:
Made a fix guys, see => https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1878

izogfif, 2025-11-08:
@PhyxionNL is it already released? Both winget and "Releases" section on GitHub show version 0.9.4 that was built over a year ago.

Blacktempel, 2025-12-31:
@izogfif there is no release yet as some other things have to be fixed before.
You can take a recent version from ["Actions" tab](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions) and use that one.

closed 2025-12-31 by Blacktempel