LibreHardwareMonitor/LibreHardwareMonitor #1688: MSI MPG X870E CARBON WIFI - No fan sensors or controllers detected

Capybara-sea, 2025-04-07:
FanControl launches successfully but does **not display any usable fan sensors or controllers**, despite the fact that LibreHardwareMonitor **correctly detects** both RPM sensors and fan control channels.

[image]

Parrallax2025, 2025-04-22:
Hi! I saw your post about the MSI MPG X870E CARBON WIFI and FanControl not detecting any fan sensors or controllers. Could I ask — how exactly are your fans connected to the motherboard?

Are you using any kind of fan hub or splitter (like DEEPCOOL FH-04 or Arctic Case Fan Hub), or are they plugged directly into the SYS_FAN / CPU_FAN headers?

I’m troubleshooting a very similar issue and trying to figure out if the fan hub might be the cause. Thanks a lot in advance!

Capybara-sea, 2025-04-23:
> 您好！我看到了您关于 MSI MPG X870E CARBON WIFI 和 FanControl 检测不到任何风扇传感器或控制器的帖子。请问一下，您的风扇具体是怎么连接到主板的？
> 
> 您是否使用任何类型的风扇集线器或分离器（例如 DEEPCOOL FH-04 或 Arctic Case Fan Hub），或者它们是否直接插入 SYS_FAN / CPU_FAN 接头？
> 
> 我正在排查一个非常类似的问题，想看看是不是风扇轮毂的问题。提前谢谢！

The AIO cooler fans, the AIO pump, and the four case fans that came with the LianLi 206 case are all plugged directly into the motherboard's PUMP_FAN and SYS_FAN headers.

Alcolawl, 2025-05-09:
X870E Carbon WiFi support was merged on April 22nd. This can be closed.

Nox7M, 2026-02-06:
Somehow, my AIO (NZXT Kraken 360 RGB) is being detected as SYS_FAN 3 in Fan Control. 
It is connected to PUMP_SYS though.
This also prevents me from controlling the fan that is actually connected to SYS_FAN 3.