LibreHardwareMonitor/LibreHardwareMonitor #1694: MSI MPG B850 Edge TI WiFi support issue

XiaoYe0565, 2025-04-14:
I cannot display the sys fan for my MSI MPG B850 Edge TI WiFi motherboard in Fan Control, nor can I control the sys fan. Is this because LHM doesn't support MSI MPG B850 Edge TI WiFi yet?

Darkling2023, 2025-04-14:
Several of the newer MSI boards are not supported yet. They are working on it. See here: https://github.com/Rem0o/FanControl.Releases/discussions/2982

Grosser-Kurfuerst, 2025-04-30:
There is still no support for the MSI MPG B850 Edge Ti wifi motherboard, and the list in https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1644 also does not include any support plan for this motherboard.
hope support for the B850 Edge Ti can be added soon. Thank you very much for your efforts.