LibreHardwareMonitor/LibreHardwareMonitor #1701: Units Changing for Transfer Speeds - is there a way to use a fixed one?

DReffects, 2025-04-23:
hello there!

 I am using LibreHardwareMonitor and it's backwards compatibility to "OpenHardwareMonitor" to monitor a bunch of machines. 

An issue arose since LHWM constantly changes the transfer speed units for example for the Ethernet Connections or Harddisks from KB/s to MB/s and back. 

Is there a way to prevent this and always use MB/s for example?

Thanks :-)

blackholeearth, 2025-04-30:
At Line 233

https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/b5553c37a2e44cf82da0e124888c1c436daf4bd9/LibreHardwareMonitor/UI/SensorNode.cs#L183-L253

--------

Add bool above the func

````csharp

// İf you want user to change this ,
/// expose this  to user gui or httpServer 

//Old  behavior is false,,
// Home assistant user want fixedMb . So true for them 
Pulic bool fixedAsMb=true; 

String ValueToString(float value )
{

// ....... 

// Change Line 233 like this;;;
if(fixedAsMb)
{
   result = $"{value / _1MB:F1} MB/s"; 
}
else
{
   result = value < _1MB ? $"{value / 1024:F1} KB/s" : $"{value / _1MB:F1} MB/s"; 
}

// ...... 
}
````

DReffects, 2025-05-01:
> `Answered via phone`..
> 
> Yes there is. Modify and rebuild..

Ah, nice! Thanks for pointing out the correct place in the code!

I have never worked with github and compiling before - can you give me a quick outline on how to proceed?

blackholeearth, 2025-05-01:
Download install  visual studio community 2022.

At installer Choose Desktop winforms in workloads.

Download this repo from main page green button.

Open with vs.  By clicking 
.sln file.

Replace the code...

Select Default project ass app.
Not the lib.

Press f5..

----

Ask the gemini 2.5  at ai studio ,  it can guide you in details..

DReffects, 2025-05-02:
> Download install visual studio community 2022.
> ...
> Ask gemini 2.5 at ai studio , it can guide you in details..

Thanks!! I'll give it a try!